keirf/flashfloppy #489: Disk size when used alongside second drive

atoone, 2021-06-08:
I have a Tatung Einstein, configured with two 3.5" external drives (0: and 1:) and an internal 3" drive (2:)

I'm using an OpenFlops drive to replace one of the external 3.5" drives, with Flash Floppy (3.23 at present).

Disk images are either 190K (3" drive) or 790K (3.5" drive).

If I install the OpenFlops unit as the only external drive (ie. drive 0:) I can boot and dir standard (190K) Tatung .DSK images, as well as being able to see the internal (drive 2:) 3" drive.

However, if I install the OpenFlops unit alongside an original 3.5" drive (ie. 3.5" drive as boot disk 0:, Open flops as drive 1: - using the jumper on S1), it reports the disk images sizes as 790K rather than 190K, and directory listings are corrupt - the disk cannot be read.

I'm really not sure if this is a hardware, software or configuration issue. The Einstein happily works with physical 3.5" and 3" discs together, but the OpenFlops unit reports the wrong disk size when plugged in at the same time as a physical 3.5" drive.

atoone, 2021-06-08:
Issue found - the System boot file on each disk defines which drives have which capacity, and does not rely on information from FlashFloppy or the OpenFlops hardware to identify disk sizes.

It would be helpful if FlashFloppy were able to change drive number in software as this would allow maximum flexibility when swapping into existing systems, particularly where disk images may change depending on drive assignment.

closed 2021-06-08 by atoone

keirf, 2021-06-08:
> It would be helpful if FlashFloppy were able to change drive number in software as this would allow maximum flexibility when swapping into existing systems, particularly where disk images may change depending on drive assignment.

Not possible. There is one pin on the MCU for drive select and it is physically assigned by jumper connecting to the appropriate floppy-header pin.

atoone, 2021-06-08:
> Not possible. There is one pin on the MCU for drive select and it is physically assigned by jumper connecting to the appropriate floppy-header pin.

I figured that was probably the case. 

Now I just need to figure out how to get the Einstein to format a disk image - getting verify errors at the moment. This is quite challenging when it's not clear where problems actually lie.

keirf, 2021-06-08:
You can run the logfile firmware and get some idea of what FlashFloppy sees being written.