keirf/flashfloppy #490: Formatting DS 80-Track disk fails on Einstein

atoone, 2021-06-10:
Tried to format (as drive 1, using an 800K image blank.tatung.dsk) - FF displays track/side apparently being written during format, then when the host tries to verify the disk, it reports "Verify Error, Track 00, Sector 00". Image file attached. 

[IMG.CFG.txt](https://github.com/keirf/FlashFloppy/files/6631766/IMG.CFG.txt)

Logging reports:

** FlashFloppy v3.25 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Apr 25 2021 15:06:36
Board: Enhanced
Config: Flash Slot 0 (ver 2, size 69)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: SDA held by slave? Fixing... Done
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 058F
 PID : 6387
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : Generic
 Product : Mass Storage
 Serial Number : CCE714B2
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 104 items
0:F: 'blank.tatung.dsk' 
Cache 103 items
Mode: Native
Current slot: 3/6
Name: 'blank.tatung' Type: dsk
Attr: 20 Clus: 0002259c Size: 819200
Fast Seek: 1 frags
Cache 33 items
Cache 103 items
Before: After: Current slot: 4/6
Name: 'diskmag1' Type: dsk
Attr: 20 Clus: 00022532 Size: 215296

atoone, 2021-06-11:
Update - the external drive connection does not use either pin 2 or 34, so these don't appear to be necessary for correct functioning. This is the pinout for the external drive socket

[image]

This appears to be the spec for the disk formats:

[image]

keirf, 2021-06-11:
I wonder what the exact track layout is, including sector IDs. It looks like FlashFloppy hasn't picked up on any expected sectors during format as it has not logged any writes.

atoone, 2021-06-11:
This is a valid 3" dsk image (40 track single side) - the only odd thing seems to be that on even number tracks (0, 2..) the sector IDs appear 'wrapped' - ie. 5,6,7,8,9,0,1,2,3,4 (If I'm reading the data right)

[SYSTEM80.zip](https://github.com/keirf/FlashFloppy/files/6641034/SYSTEM80.zip)

If I can do anything to provide additional information, let me know. I really appreciate your time on this.

keirf, 2021-06-13:
So can you read the DSK file but not:  Modify (write sectors) on it? Format it? I agree it looks like a pretty straightforward DSK file. The skew is a bit unusual, rotating the sector IDs on some cylinders, but shouldn't be a problem.

keirf, 2021-06-13:
Perhaps try a DD HFE image file? https://github.com/keirf/FF_Images/tree/master/Empty/HFE

retrogamer1024, 2021-07-02:
I too have had some issues with DOS80.

As suggested by Keir, I can confirm the empty DD HFE image works fine with DOS80 on the Einstein. BACK80 happily formats the blank as double sided with 80 tracks per side, the image operates as expected and an inspection with the track analyser of HxCFloppyEmulator shows no errors.

The standard Einstein format (under  XtalDOS 1.31) is single sided (i.e. 1 head), 40 tracks, 10 sectors per track numbered from 0 to 9 with 512 byte sectors. That's 200k [204800 total size]. 

I've created custom images in HxCFloppyEmulator (0xE5 is the format value by the way) and exported 40 track standard images as .HFE, .DSK and raw .IMG with an IMG.CFG as above that all work fine with FlashFloppy with XtalDOS 1.31 saving, loading and running files. I reformatted them using BACKUP on the Einstein and all worked as expected with formatting, verifying the format and verifying and running newly saved files.

Remembering to run CONFIG to set the Gotek as single sided 40 track, I tried the same 40 track files with DOS80. I reformatted using BACK80 and all operate as expected. Worth noting that the formats in DOS80 are CP/M 2.2 compliant with a Disc Parameter Block for each format (there are 4 selectable formats in total) that set the required values for the number of 128 byte records per track, block size (i.e. 4), block mask, etc.

I initially wrongly assumed that each head expected to see the same standard Einstein sector IDs in DOS80, and consequently none of my attempts to generate a double-sided 80 track image worked. I tried running a scan in SAMdisk on the Einstein formatted HFE image but just got a red 'Error: unsupported data encoding (FF) on cyl 0 head 0'. I did eventually ascertain the actual sector numbering when running a scan in SAMdisk on a converted DSK generated from this HFE formatted by BACK80 on the Einstein. It would seem, by the way, that converting from an Einstein formatted HFE blank appears to be the easiest and most reliable way to generate a suitable DSK for DOS80.

The scan confirmed the double sided 80 track DOS80 format has 20 sectors per track with 512 byte sectors numbered from 0 to 9 on head 0 and numbered from 10 to 19 on head 1. That's 800k [819200]. Custom generated images need the 'inter side sector numbering' option ticked to ensure different sector numbers for each head (otherwise created images appear to format but do not verify - see the issue originally identified above). Worth pointing out that the format value needs to be set at 229 (=0xE5) or a directory error gets reported when trying to save a file. I note there is also some sector skewing, but replicating this does not seem critical. 

I've now generated 80 track .DSK, and .HFE images with the correct parameters that seem to work as expected with DOS80. Great!

But using the raw .IMG image reports a 'No Sector 1' error on the Einstein. I presume it needs IMG.CFG to indicate the track sectoring but I'm not sure how to indicate different sector numbers for each head. Any pointers gratefully received ...

Keep up the brilliant work, Keir!

keirf, 2021-07-02:
Attach an HFE formatted with the required sectoring, and perhaps an equivalent IMG file if you can, and I can post an IMG.CFG stanza.

retrogamer1024, 2021-07-02:
Many thanks, and happy to report that all images now work brilliantly with FlashFloppy! 

Should have examined the example .CFG files more closely as the Kaypro example is so close to what's needed - apologies for not spotting this earlier! 

For Einstein users who need it, the image configuration is:

#Tatung Einstein DOS80 double sided 80 track 800k
[::819200]
cyls = 80
heads = 2
secs = 10
bps = 512
tracks = 0-79.0
  id = 0
tracks = 0-79.1
  id = 10

Given the detailed explanation above to support Einstein users, I would think this ticket can now be closed.

atoone, 2021-07-05:
Thank you both for your help on this - the hfe image works, without the config file above. If I'm using that config, do I need a specific blank disk image file?

retrogamer1024, 2021-07-05:
Great. I've posted extensive details on the Tatung Einstein Forum here:

https://www.robheaton.co.uk/tatung/viewtopic.php?f=3&t=83

Hopefully this will answer any further questions.