keirf/flashfloppy #494: Roland MC-500 wont load Bootdisk.

andr229k, 2021-06-16:
T: Roland MC-500 installed with Floppy Emulator with flashfloppy-firmware 3.25.

settings as per following [Reference: https://github.com/keirf/FlashFloppy/issues/97]
Jumper = S0
Interface = JC

S:
Switching on Roland MC-500 and it initiate the booting process and after a while it freezes. Booting was not complete and all buttons blink as per video:
https://drive.google.com/file/d/1a0lF-uaVwuWtbbmpkTnyGb9d9DWHA9pW/view

D: N/A

A:
* Tested another usb-stick, no changes.
* Tried to flip IDE-cable, it tells me to flip cable.
[SuperMRC v2.00.zip](https://github.com/keirf/FlashFloppy/files/6663709/SuperMRC.v2.00.zip)

Boot disk:
[SuperMRC v2.00.zip](https://github.com/keirf/FlashFloppy/files/6663709/SuperMRC.v2.00.zip)

Any suggestions?

andr229k, 2021-06-17:
FFLOG attached:
[FFLOG [Roland MC].TXT](https://github.com/keirf/FlashFloppy/files/6671315/FFLOG.Roland.MC.TXT)

andr229k, 2021-06-20:
[FFLOG.txt](https://github.com/keirf/FlashFloppy/files/6682136/FFLOG.txt)
New log

keirf, 2021-06-20:
None of these have anything suspect in them. Are you running until the machine freezes? Then eject the mounted image (eg re-enter navigation mode on the Gotek), then pull the USB drive?

andr229k, 2021-06-20:
> None of these have anything suspect in them. Are you running until the machine freezes? Then eject the mounted image (eg re-enter navigation mode on the Gotek), then pull the USB drive?

The Roland MC-500 was running till it froze after approx. 40 sec.

Facts:
•	Roland MC-500 boots fine with the old original drive.
•	Roland MC-500 crashes after a certain number of seconds [approx. 40 sec], if no USB-stick is inserted.
•	Roland MC-500 initiate the booting from USB-stick, but crashes after a certain number of seconds. [approx. 40 sec.]

Crash = both screens shut down and all leds flicker. It only remains to switch off the MC-500

Example of crash without USB-stick inserted:
https://drive.google.com/file/d/1jDugQxSI-jlJQJM64G9eWy_BGfapE9Oq/view?usp=sharing

USB-stick content:
[USB-stick content.zip](https://github.com/keirf/FlashFloppy/files/6682457/USB-stick.content.zip)

andr229k, 2021-06-21:
Pulled all apart and put together, now it works normal.

May a problem with IDE cable.

closed 2021-06-21 by andr229k