keirf/flashfloppy #495: New Gotek Systen design - JA / Black buttons issue

andr229k, 2021-06-22:
There is a new design revision being pushed out recently, which has the small chip but the 6 pins as the older design for Rotary and OLED. This is probably the forerunner to the new pre-assembled Gotek with OLED and Rotary.

**I observed:**
•	JA and the two black buttons have no functionality anymore. They may have been re-mapped? [which means that you cannot eject or update from the USB]
•	Rotary can be connected again directly on the 2 pins beside OED, like before.

**Pictures:**
[image]
[image]

RetroClinic, 2021-06-22:
Hmm, I too received a batch of these new ones yesterday, and will try one in the next day or so. Is it possible the encoder button is now mapped to one or more of the new 6 pin header the factory encoder connects to? Or will a special mapping be needed for this one?

keirf, 2021-06-22:
I received this board from factory today and I will have a firmware for test tomorrow. It will need test across all Goteks: old 64-pin stm32, original 48- and 64-pin Artery boards, and this new board.

My board is part of the new factory fitted OLED and rotary encoder range. Latter uses the new six pin header but this is optional as the old rotary encoder header is available again on this new PCB.

andr229k, 2021-06-22:
> Hmm, I too received a batch of these new ones yesterday, and will try one in the next day or so. Is it possible the encoder button is now mapped to one or more of the new 6 pin header the factory encoder connects to? Or will a special mapping be needed for this one?

I also had another strange observation. My Rotary is only able to turn around, if ground from Rotary is connected to JA or the following pin:
[image]

keirf, 2021-06-23:
> I also had another strange observation. My Rotary is only able to turn around, if ground from Rotary is connected to JA or the following pin:

Well your rotary does need to be grounded, right? And you just mentioned two ground pins. So that makes sense.

keirf, 2021-06-23:
Okay here is a test firmware, feel free to give it a go on any Gotek (original, 48-pin Artery, 64-pin Artery, and this new SFRKC30 board).

Instructions: Download the link, unzip, then unzip the non-debug zip inside. Use the HEX or UPD file within as per usual.

https://github.com/keirf/FlashFloppy/suites/3063891179/artifacts/69751002

keirf, 2021-06-23:
@RetroClinic you can use above firmware and hook up a rotary encoder in the "traditional" location.

RetroClinic, 2021-06-23:
Great! Put that on one of the new ones, and the buttons are back in business. I've not tested it on the earlier boards, I'll keep the regular 3.25 on those as we know that works on them.

I'll finish the build later on today, install the encoder and give it a test on the Beeb. Thanks :-)

closed 2021-06-24 by keirf