keirf/flashfloppy #496: 2.88MB ED support for x86 PC

cyberluke, 2021-06-26:
Hi, I have Pentium 1 133Mhz MMX (motherboard FIC PA-2005). 

FlashFloppy is running firmware 3.25.

BIOS set to 2.88MB drive.

RAW image for 2.88 created using MagicISO.

FlashFloppy successfully reads and loads drive (if the size is over spec FlashFloppy will show error while loading RAW image from USB).

When trying to list the content of this drive on computer, it shows error: Drive not ready Abort/Retry/Fail

Is there some newer firmware or some additional configuration for IBM PC 2.88MB drive? Do I need to set up some additional pins?

I did read that some modern FDC controllers will not support this, but Pentium 1 should support. It was made in this era where you can have more floppy drives. Modern MBs support only one floppy drive.

I have ISA Floppy Drive controller for 4 floppies that claims 2.88MB support, so I will try that as well.

cyberluke, 2021-06-26:
EDIT: The error is more precisely General failure

cyberluke, 2021-07-01:
I did not have any luck yet. Will re-install OS on another machine and give one more try with a dedicated IDE controller that has 2.88MB support.

keirf, 2021-07-01:
Is this an older STM based Gotek?

cyberluke, 2021-07-01:
I bought it modded in Germany on E-Bay last month. This week I got a second unit. Both will be the classic Gotek before they changed the chip (the recent firmware changes).

keirf, 2021-07-01:
My understanding is the 2.88MB support is not as broad as you might think. It was never a popular option of course.

cyberluke, 2021-07-01:
Yes thats why I bought 80 USD floppy controller. Look how many different 2.88mb drives it can handle. Can you tell me what type did you implement?
[image]

H
L
or
N
[image]

keirf, 2021-07-01:
I think H/L/N probably indicates how the drive gets put into 2.88MB mode, via pin signal High Low or None. Since FF autodetects 2.88M basedc on image size, the correct answer in that case would be N.

cyberluke, 2021-07-01:
Ok, perfect. Thank you. I will try it today or tomorrow. I know people use FlashFloppy for musical equipment (Akai, Roland, Yamaha, Korg). But I read somewhere that on PC it needs a different signaling and I would say that there is a high chance the control codes are not being sent properly on IBM PC compatible machine. There is also IBM PS/1 and IBM PS/2 2.88MB floppy drives which use different cable and are not compatible with IBM PC (you can find these at lower price on E-Bay). So I am 100% sure this ISA controller (GSI Model 21) will work with 2.88MB floppy drive, so this time it must work with FlashFloppy. It would be great to have more individual configuration for 2.88MB in ff.cfg. Autodetection might cause trouble to some PC onboard floppy controllers, perhaps. Or if image size is higher than 1.44MB, but smaller than 2.88MB - what will it detect?

cyberluke, 2021-07-02:
I'm back. Finally did test all combinations. This new GSI Model 21 floppy/IDE/CD-ROM controller has two floppy controllers. One standard, second is specifically for 2.88MB and it is written on the PCB next to it.

FlashFloppy is set to IBM PC mode using two jumpers.

Controller settings:
option 2.88 H:
- 1.44MB image does NOT read
- 2.88MB image does NOT read
option 2.88 L:
- 1.44MB image does NOT read
- 2.88MB image does NOT read
option 2.88 N:
- 1.44MB image DOES read
- 2.88MB image does NOT read

1) I have official 2.88MB floppy image of installer. Second image is created by me. Tested both.
2) The behavior is exactly same as on P133 with onboard controller. When set to 2.88MB, only 1.44MB images are loaded.

With 2.88 L or 2.88 H mode, your autodetection does not work and it does not read even 1.44MB images.

My belief is that you did not test 2.88M images by loading them. You did test only if the drive will be found in OS. Or you did test 2.88M, but it was on Amiga or other non IBM PC system. Therefore in my humble opinion you need to add a specific IBM PC system support.

When loading 2.88MB image, FlashFloppy will make different seek with sound for 2.88 H/L and 2.88 N modes. So the behavior is different in some way. For 2.88 N I see it tries to load the track and it jumps from T:00.0 to T:02.0 in a loop. It cannot get past that sector. It will always reset reading to T:00.0. That is why it displays error.

cyberluke, 2021-07-02:
PS: image type is IMG, IMA, DSK (IBM MFM Raw Sector)

keirf, 2021-07-02:
I've never tested ED support on a PC and I'm rather unlikely ever to hunt down an ED controller and do so. There has been one positive report of success in issue ticket #294 with an older version of FlashFloppy (v3.7a).

cyberluke, 2021-07-02:
Ok, I will try to downgrade. I tried the latest firmware now (3.27) plus I tried tweaking various settings such as host and interface. It did not help.

Interesting is that pin02 = auto* | nc | low | high | rdy | nrdy | dens | ndens | chg | nchg

This pin02 should be responsible for setting the type 2.88 H / 2.88 L / 2.88 N. If I set auto or nc and 2.88 N it reads at least 1.44M. If I set pin02=high and BIOS controller to 2.88 H, it does not help and does not work even for 1.44M.

cyberluke, 2021-07-02:
v3.7a will display garbage on OLED display (I have knob + OLED + PC Speaker mod)

cyberluke, 2021-07-03:
"I've never tested ED support on a PC" - yes and I have read that signaling for ED on PC is different than on Amiga. That would support other user findings there.

Here is IBM PS/2 floppy, different pins, no power connector: https://www.ebay.com/itm/144097330149

Here is another, regular, floppy with power connector: https://www.ebay.com/itm/164831254434

If you can compile 3.7a with OLED support, I can test it.

keirf, 2021-07-03:
Amiga doesn't support ED. It's pretty rare overall. I would have to take a look at v3.7a as the OLED failure is unexpected, I don't remember having fixed or added further low level OLED support since then. I'm away from Monday for two weeks.

cyberluke, 2021-07-03:
v3.7a - display is showing random pixels like it is connected to other pin. I put image.cfg and default mount image to 2.88M. It does not load. Other 1.44M also does not load. So as a blind test, I guess v3.7a has some different pin connection. Even I supply ff.cfg.

`interface=ibmpc
host=unspecified
pin02=nc
track-change=instant
write-drain=eot
motor-delay=0
display-probe-ms=10000
autoselect-file-secs=0
autoselect-folder-secs=0
folder-sort=never
nav-mode=native
oled-contrast=64
display-off-secs=120
display-scroll-rate=150
step-volume=20
display-type=auto
oled-font=6x13`

We can fix it, I'm also software developer. But this is not my project and I don't understand the theory.

I can buy now real 2.88M floppy for $70 and will have it delivered in 3 weeks and I can try all my PCs and this GSI 21 controller with real floppy and that can finally point to where is the issue. But this controller sold back in the day for $100, so it is high quality one.

keirf, 2021-07-03:
Okay, let's see. There are few changes to OLED since v3.7a, I can do a `git log src/display/lcd.c` and see nothing major there. It could be worth restting cached config after the downgrade, by holding both buttons for a few seconds (or pressing the rotary encoder for a few seconds) with no disk inserted.

Regarding disk capacity detection, the problem with getting this working on PC is that the interface was never standardised. You will find this Application Note interesting (especially from page 9: "Interface between [FDC] and the drive"). It may give you some ideas. http://www.pix.net/languard/pdfs/29209302.pdf

cyberluke, 2021-07-03:
Yes, prior to opening a thread here, I did search here and Google as well. I have seen this PDF. So I was thinking this is already implemented / taken care of in FlashFloppy.

I have not found any disk capacity profile for 2.88M. Only some HxC cfg for 2.4M. I see in the code it checks for number of sectors, not image size.

It's not that the drive would give some fail or something. I just refuse to move track above 02 and it is jumping between 00 and 02. So my guess is it needs send some information to PC in a different way, so it won't request the same data over and over again and then give general floppy fail.

keirf, 2021-07-03:
The various pin interfaces cannot be handled entirely by FlashFloppy on Gotek. As various pins are simply not controllable.

keirf, 2021-07-03:
Could it also be the case that the system BIOS must support 2.88M?

cyberluke, 2021-07-03:
No, P1 133Mhz MMX has bios support and it is same situation with this ISA
controller. And because controller has its own BIOS it can add support for
anything. It also has support for larger drive which bios does not support
and it is working. Plus I have additional XT IDE bios. I am using 32GB SD
card over IDE on this machine right now with Windows 95, even bios never
supported those.

Right now I switched to 2.88 H from 2.88 N on the controller and I'm trying
various pin02 and pin34 combinations (dens, ndens). Together with
ibmpc-hdout interface.

keirf, 2021-07-03:
There may also be control via pins 6 or 33? It's unclear how common it is that these are drive outputs vs inputs!

cyberluke, 2021-07-03:
Yes, I think I will need to modify the code and build it myself to add
support for pin 6 configuration. That should be easy.

I see the biggest problem is different gap size, but I believe this is
already handled in FlashFloppy (for 36 sector images). Did not check it
though.

keirf, 2021-07-03:
You know it could be the GAP2. I considered that it only matters that it is made larger, for real physical sector writes to a real disk. But see the Application Note, page 6, paragraph 2: "On the read back by the 82077AA/SL..."
It implies that the PLL is only primed to scan for the sync mark after 43 bytes in 4MB mode. 

With correct GAP2 = 41 bytes, this means that PLL sync starts safely 2 bytes into the 00 presync pattern.

With FlashFloppy's default GAP2 = 22 bytes, this means that PLL sync starts a few bytes after the A1A1A1 sync pattern. Disaster!

keirf, 2021-07-03:
Try this test firmware which should change the ED/4MB GAP2 to 41 bytes:

https://github.com/keirf/FlashFloppy/suites/3149733163/artifacts/72410125

(you need to be logged into Github for the link to work)

cyberluke, 2021-07-08:
Perfect, I will test it today.

cyberluke, 2021-07-08:
WOW! IT IS WORKING, thank you! You are THE KING :-D

See, I smell something bad. Now everyone can use FlashFloppy with 2.88MB images, so we don't need to split some installation media, for example zip disk I had to manually split to two images and then manually copy to C drive.

I have just tested the 2.88 N type and it reads both 1.44MB and 2.88MB. Can read content, can read file.

keirf, 2021-07-08:
Great I need to clean this up and also make GAP2 configurable in IMG.CFG. Then it can be rolled into next release.

cyberluke, 2021-07-09:
Tested new firmware with Pentium 133Mhz MMX and onboard IDE controller (BIOS shows 2.88MB on both A: and B:). WORKING!

lippone, 2021-07-10:
6bxc formatting are not working

keirf, 2021-08-07:
ED formatting is unreliable even on STM-based Goteks (with the full 64kB RAM). Possibly you will find this better on a build of the v4 pre-release firmware. It may get fixed in future on v3 branch too, but for now it's a known issue I'm afraid.

keirf, 2021-08-10:
@lippone You may want to try the new v4.0a pre-release. This contains the GAP2 fix, plus the new async-I/O data path which hopefully better deals with ED track formatting.

lippone, 2021-09-20:
There are still som bugs on v4.0a when use sys a: & format 

[FFLOG (2).TXT](http
[image]
s://github.com/keirf/FlashFloppy/files/7193385/FFLOG.2.TXT
[image]
)
[image]
[image]

closed 2021-11-24 by keirf