keirf/flashfloppy #497: Wifi support over USB

cyberluke, 2021-06-26:
I am currently running MicroSD in USB adapter in FlashFloppy drive.

I have seen some USB flash with wifi support (Sandisk, etc) as well as SD cards with wifi.

Will they work?

The common issue is that, for example, HxC floppy emulators will turn off USB flash after reading file or disk operation. This means voltage is cut, which means Wifi will disconnect. How is it here? Is there some option for permanent voltage delivery? Perhaps disabling OLED screen turn off after X seconds?

keirf, 2021-06-26:
USB is permanently connected to the floppy power 5v input. So power is never and can never be cut.

cyberluke, 2021-06-26:
Perfect, thank you! So Wifi streaming to USB flash should be possible :-)

cyberluke, 2021-06-26:
I have ordered 8GB SD Wifi card (size for compatibility purposes) including CF card adapter. Now I need to get USB Flash SD card adapter and it should work. I will test and verify in 3-4 weeks.

Tolik-Trek, 2021-10-10:
> I have ordered 8GB SD Wifi card (size for compatibility purposes) including CF card adapter. Now I need to get USB Flash SD card adapter and it should work. I will test and verify in 3-4 weeks.
> 
> https://www.aliexpress.com/item/33057103321.html

Is it work?

cyberluke, 2021-10-10:
The one I bought is a bit tricky because of proprietary communication protocol. Their app is outdated. They have only Android app. No website UI is running there. Some other card could work. This one could work as well and I will get back to it next year. I have decompiled the Android app, so I see the Java communication protocol. It sends some messages over UDP. I would need to create some desktop app, where you can upload files using this protocol. Then it would work.

So technically it is working as of now. The only caveat is: cannot transfer files from PC. Only from Android app. Chinese always lacking in software department. I bought a lot of hardware for my startup from Shenzhen past 5 years. Always problem with documentation and software, zero support, zero updates, bad software quality. But I can do it myself.

cyberluke, 2021-10-10:
The website is down: http://web.archive.org/web/20191003193622/http://voorca.net/ ...there should be the drivers. I have contacted AliExpress seller again to provide drivers.

Tolik-Trek, 2022-01-20:
> The website is down: http://web.archive.org/web/20191003193622/http://voorca.net/ ...there should be the drivers. I have contacted AliExpress seller again to provide drivers.

Hi. Found something?

cyberluke, 2022-01-20:
No, only decompiled old Android mobile software. Im busy, would have to pay
someone to develop a software for PC. It will take me probably 6 months to
solve it.

Tolik-Trek, 2022-01-20:
> No, only decompiled old Android mobile software. Im busy, would have to pay
> someone to develop a software for PC. It will take me probably 6 months to
> solve it.

Toshiba FlashAir W-03 can work nice

cyberluke, 2022-07-06:
I’m back, so Toshiba works? I’m gonna buy it next month.