keirf/flashfloppy #500: Display continually scrolling through directories

baileyt3, 2021-06-30:
I installed 3.26 on a GOEX, the display scrolls/flickers through all directories continuously, pressing or rotating the wheel had no effect, installed 3.25 and the problem stopped, working normally.

keirf, 2021-06-30:
I will test on the only GOEX board I have, which I think is r1. Do you have an r2 board (it will say on it, after LT.OSD).

keirf, 2021-06-30:
Okay it is a bug in handling "Enhanced Gotek" models. Please try the following build. Use the non-debug zip file contained within and apply the UPD file as usual.

https://nightly.link/keirf/FlashFloppy/actions/artifacts/71644243.zip

baileyt3, 2021-06-30:
Hi, just had a look and mine has r3 on it.

iancourt, 2021-06-30:
> https://nightly.link/keirf/FlashFloppy/actions/artifacts/71644243.zip

Confirming that this build fixed the issue for me on my OpenFlops, flashed in DFU mode.

keirf, 2021-06-30:
Thanks I need to test it across other goteks and then make a release

keirf, 2021-07-03:
Fixed in v3.27

closed 2021-07-03 by keirf