keirf/flashfloppy #501: RDY signal on PIN 34, DCHG signal on pin 2 by default or with a jumper

Wierzbowsky, 2021-07-02:
I have quite a problem with Sony MSXs (F900 model and similar) that require RDY signal on pin 34 and DCHG signal on pin 2. Without this Sony machines don't boot if USB drive is not inserted. 

Would it be possible to either create a firmware branch that has RDY on pin 32 and DCHG on pin 2 by default (on power on) or switch to this mode with one of the jumpers (JA, JB, JC or J5). Thanks in advance.

keirf, 2021-07-02:
RDY on pin 34 and DCHG on pin 2 is default out of the box behaviour.

You mention RDY on pin 32 in your second paragraph but I assume this is a typo?

Wierzbowsky, 2021-07-03:
Yes, that was a typo. If the RDY on 34 and DCHG on 2 is the default behavior, then this ticket is not needed.

closed 2021-07-03 by Wierzbowsky