keirf/flashfloppy #504: IMG.CFG: Custom entries only affect .IMG files and not .DSK files.

stephen-usher, 2021-07-07:
When a custom format is created in IMG.CFG, e.g.

_[apr]
cyls = 80
heads = 2
secs = 9
bps = 512
rate = 500
rpm = 600_

Raw image files with the name filename.apr.img work correctly however DSK format files with names such as filename.apr.dsk do not pick up the settings. In this case the host, expecting a 600 rpm drive and 500bps data rate gets the data at 250bps.

The documentation states that .img, ima and .dsk files will all be affected.

keirf, 2021-07-07:
I will investigate this in a week or so.

keirf, 2021-07-19:
I'm now back home and will make this a priority.

keirf, 2021-07-19:
Confirmed that two files *foo.apr.img* and *bar.apr.dsk* both get correctly mapped to the *[apr]* tag.

stephen-usher, 2021-07-19:
I did take a quick look at the .DSK source code and I couldn't see where it was using the "rate" being set by the configuration. It's certainly not sending the data at 500bps to the controller, always 250bps, unlike IMG files. (FF v3.26) I can't remember the rpm being referenced either.

keirf, 2021-07-19:
Oh I see.. CPC EDSK files are always assumed to be double density data rate. They are unrelated to the IMG handler. However a DSK file which is not CPC EDSK format falls back to the IMG handler as some systems use the DSK suffix for raw/IMG files. And in that case of course IMG.CFG is applied.

I doubt I will mess with the EDSK handler. It took a lot of work to get it working with most CPC and Spectrum images. And those are the systems it's intended for I believe.

keirf, 2021-08-22:
Before I close this down, do you have any EDSK images which use the rate+encoding extensions described here: https://comp.sys.sinclair.narkive.com/ZunCAuv8/dsk-edsk-file-format-data-rate

I will not auto-detect other data rates and encodings for EDSK images, but if there is demand I could handle explicitly requested rates and encodings within the track headers.

keirf, 2021-09-10:
No further activity so closing for now. Can be reopened if need be.

closed 2021-09-10 by keirf