keirf/flashfloppy #505: IMG.CFG: New default values using empty tagname require number of heads.

stephen-usher, 2021-07-07:
The empty tagname setting in IMG.CFG mandates the specification of the number of heads. This means that when specifying a different disk rotation speed and bit rate either single-sided or double-sided images will no-longer work.

It could be argued that the 'rpm' value should not actually be in IMG.CFG at all as it's a property of the drive/disk controller and independent of the image and hence should be a global setting in FF.CFG instead.

keirf, 2021-07-07:
In latest firmware you can have multiple empty tagnames discriminated by file size. This should suffice?

stephen-usher, 2021-07-08:
It's not perfect as (I thnk) .DSK images aren't necessarily all the same size if they don't store empty tracks.

You might as well just specify a complete set of new image types and rename the files.

I don't know how difficult it would be to implement but could the NULL tag "merely" modify the global default values so that you only have to specify the differences from them?

keirf, 2021-07-19:
If you can't discriminate by name (because you want NULL tag) nor by size (since partial images aren't padded) then I'm not sure how you envisage this working. Those are the two keys by which a full IMG config is plucked from IMG.CFG.

stephen-usher, 2021-07-19:
Well, if a floppy disk controller will only expect drives with a certain RPM and/or bit rate then you want to globally modify the default values for all recognised disk images.

i.e. change all disk images or whatever size to say, 600 rpm and 500 bps instead of the default 300 rpm and 250 bps. The disk type detection wouldn't change from the default.

In this example it's purely a matter of what the controller expects from the device and has nothing to do with the image format itself.

Really, the RPM should not be image specific as that's at the device/controller level and not a property of the medium. The bps value should be scaled by the rpm ideally too (as it's really a bits-per-mm * mm-per-second).

keirf, 2021-07-19:
Some drives change RPM by track (eg Mac CLV). Some drives change by disk type (eg. Amiga HD vs DD disks, PC 5.25 drive HD vs DD disks). It's not necessarily a constant for the drive.

Also bit rate is usually truly bps: eg. if the disk spins at 360RPM the bitrate remains 250kbps or 500kbps and the bits just end up physically further apart around the disk.

closed 2021-08-22 by keirf