keirf/flashfloppy #508: QuickDisk Track Location / Status indicator?

DOSputin, 2021-07-09:
With QD in Akai X7000,  
track indication always says T:00:0.   OLED does not give visual indication that read/write is occurring. 
Is there something I'm missing?

keirf, 2021-07-10:
Quickdisk is arranged as a single spiral track. It is endlessly read in a loop while the motor is on. Writes should be indicated on OLED with a W character.

The OLED could be somewhat prettified for QD in future, but it's behaving as it should do currently.

DOSputin, 2021-07-10:
The batch of drives that I bought came without LEDs installed.  Have now installed the green LED, and that illuminates when the drive is doing i/o.   There was no 'W' on the OLED.   No change to the OLED when reads or writes happen.

keirf, 2021-07-11:
Okay yes maybe the W is not implemented on QD firmware. This ticket could track that work.

DOSputin, 2021-07-12:
First, I apologize for the misinformation.  Due to my shit eyes and inattentiveness, I missed the 'W' during previous writes.  Possibly as I'm working with very tiny samples.  It's a quick process.  Today I temporarily mounted a larger OLED, that is relative in size to the stock LCD on the x7000, and it is MUCH easier to see the W now.   So yes.  There IS a 'W' displayed during write.  Nothing during reads.   The 'W' seems sufficient.  Maybe simply an 'R' during a read event?   I'm not sure how the display routines could be improved much beyond that?

DOSputin, 2021-07-12:
I need to read more into the engineering documents floating around on the nets for the QD.  Perhaps you have some?
What I'm wondering is:  Does the host have a way of getting status from the QD?  Due to the unique single-track spirally nature of the Quick disk, maybe not?  Is is possible to store multiple files in a track on the QD?  
I'll do some more reading and research on my own.  If you have any info and the time to help elucidate please share.

keirf, 2021-07-19:
Take a look at links on the original QD support ticket: https://github.com/keirf/FlashFloppy/issues/203

closed 2021-08-07 by keirf