keirf/flashfloppy #509: QuickDisk emulation: Roland S-220 reads QD images, but do not writes

OlafSvendson, 2021-07-10:
Hi, my first message here, just registered for.

My congrats for so nice FlashFloppy software. I installed it on several machines and it worked perfectly on a variety of musical instruments equipment.

About the QD emulation, I have FF working perfectly on the Akai MD280 (drive for the Akai S612). 

Not the same luck with Roland S-220 rack sampler. It reads the QD images, but I can´t write info on QD images. In the past, it was the same with Roland S-10 keyboard sampler (which is very related to the S-220).

As I could make FF works on the Akai MD280, I think it is not a thing of configuration or cabling. And on the S-220 the read function works properly.

Any suggestion?
thanks in advance,
Olaf Svendson

keirf, 2021-07-10:
Have you tried with and without jumper JC?

OlafSvendson, 2021-07-10:
Tried both options; JC jumpered and unjumpered. When trying to save data to a QD image, it says: **"verify error"**. On S-220 manual, page 71, it says: "QD is damaged. Replace it with a proper one".

OlafSvendson, 2021-07-10:
Update: if I use the function of saving without verification (according to manual: press F1 + save button), then the "w" appears on the gotek oled display, one blink, and dissapear. The S220 says "verify error", but it effectively (and fortunately!) stored the data, as it can be loaded afterwards.

keirf, 2021-07-10:
Are you using latest FF release? It's probably a timing thing: delay after write before FF starts reading again.

OlafSvendson, 2021-07-10:
I don´t know if it may help (and sorry if it is unpolite somehow), but it seems that this kind of problem was solved on the hxc side some time ago. I did not dig a lot into hxc, but it looks like there are some mods on the firmware (the mysterious Roland upd variant) to avoid the problem

OlafSvendson, 2021-07-10:
> Are you using latest FF release? It's probably a timing thing: delay after write before FF starts reading again.

3.27, upgraded today.

OlafSvendson, 2021-07-10:
> I don´t know if it may help (and sorry if it is unpolite somehow), but it seems that this kind of problem was solved on the hxc side some time ago. I did not dig a lot into hxc, but it looks like there are some mods on the firmware (the mysterious Roland upd variant) to avoid the problem

I got this info here: https://torlus.com/floppy/forum/viewtopic.php?t=773&start=450

keirf, 2021-07-10:
The Roland upd equivalent for FF is using the JC jumper. So perhaps it's something else here.

OlafSvendson, 2021-07-10:
Issue solved, it was my fault. Sorry for any inconvenience.

I checked all the requirements, and found that I made a mistake by keeping the S0 jumper on. :/

After removing the S0 jumper, now it works flawlessly, both reading and saving to QD image.

So: jumper present at MO and JC; correct wiring and it runs perfectly on Roland S-220

Thankyou very much for help, and keep on develop!
Olaf Svendson

closed 2021-07-10 by OlafSvendson