keirf/flashfloppy #511: SFR1M44-U100LQD GOTEK with integrated OLED orientation is upside-down

aperezbios, 2021-07-15:
GOTEK's new Artery-based SFR1M44-U100LQD includes an OLED display and rotary knob, from the factory. I'm able to flash firmware version 3.27 to this board without issue, however the orientation of the OLED display is inverted 180 degrees when I boot with the 3.27 firmware. Is this expected? Is it possible to build the FlashFloppy firmware with the orientation flipped, such that it's not necessary to correct the orientation in ff.cfg? Perhaps there's some way to detect this particular variant of GOTEK at runtime, and its OLED, and behave correctly on boot-up? I'd be happy to supply you with a SFR1M44-U100LQD for testing, @keirf

keirf, 2021-07-15:
Put this in FF.CFG `display-type=oled-128x64-rotate`

This will be cached so you don't have to keep the FF.CFG. It will only be forgotten on reset of flash configuration or if overridden by a new display-type line.

aperezbios, 2021-07-15:
Great, thanks a bunch

closed 2021-07-19 by keirf