keirf/flashfloppy #513: M0 inconsistency on latest PCB; FF v3.26/3.27

tizzulo, 2021-07-30:
It seems my two devices which like to have the "M0" jumper set need tweaks to the FF.CFG for the latest PCB "SFRKC30.AT4.7". I am not sure if this is user error or an inconsistency between PCB's. 

SFRC922D - jumper M0, interface = jppc, IMG file disk load/save/format OK
SFRC922AT3 - jumper M0, interface = jppc, IMG file disk load/save/format works sometimes
SFRKC30.AT4.7 - jumper M0, **interface = ibmpc**, IMG disk load/save OK, format works so far, not tried enough times yet to see if it's perfectly reliable

I'm not sure why - I was at a loss as to why the latest model wasn't working with my old config until someone with the same device mentioned to me they tried with interface "ibmpc" and suddenly everything works.

Thing is, on SFRC922D and SFRC922AT3, i must keep interface on jppc, as ibmpc isn't working. The 922D is on 3.27 and the 922AT3 is on 3.23.

Does anyone have a device which requires M0 and verify this? I notice the difference in IBMPC and JPPC is RDY vs DSKCHG,  
but I haven't tried messing with forcing the signal on pin 34 yet to see if I can have a config that is consistent across all 3 PCB's.

I will update as i continue to do testing.

keirf, 2021-07-30:
The only difference between jppc and ibmpc is behaviour of pin 34:
- jppc pin 34 is RDY
- ibmpc pin 34 is DSKCHG
That is all. I have verified that pin 34 works the same (is wired the same) on AT4.7 as on earlier PCBs. So nothing should differ here from a PCB perspective.

tizzulo, 2021-07-30:
Ah geez, I was hoping something about setting the jumper MO caused those to behave differently, now I am even more confused. I did find a third device which needs MO + S0, and is working with ibmpc-hdout.

I suppose I will chalk it up to user error or get a logic probe and see why my config from 3.23 stopped working

closed 2021-08-07 by keirf

tizzulo, 2021-09-09:
Just following up on this. All devices which were having inconsistencies are resolved by putting a jumper on M0 and S0 instead of just M0.