keirf/flashfloppy #514: Create empty/blank image on USB stick

roses2uk, 2021-07-31:
I'm sure this must have been raised before, but I'm blowed if I can find anything in this forum using the search function. So I apologise in advance for wasting your time.

How do I create a blank or empty image on my Gotek's USB stick so that I can save the contents of my sampler's memory as if to a fresh floppy disk? I'm especially referring to a Roland S760 rackmount and a Roland W30 keyboard workstation.

Thx

keirf, 2021-07-31:
Have a blank image ready to go and then Clone it or Copy and Paste it via the eject menu (available only if you have an OLED display)

roses2uk, 2021-08-01:
Yeah, I have an OLED display, and have seen those commands. But I don't know how to create a blank image to start with. Any 760/W30 owners care to help? Could I just not overwrite one of the existing images on the USB stick (as I have backups on my PC)?

keirf, 2021-08-02:
Yes that would work

closed 2021-08-07 by keirf