keirf/flashfloppy #515: Feature Request: Trim DSKAnnnn from OLED display

sbelectronics, 2021-08-01:
I've set up my sdcard in index mode because I like to be able to assign numbers to the files and to have those numbers shown in the status line. As such the DSKAnnnn number in the image name is redundant and causes needless horizontal scrolling of the image name on the OLED display. Would it be possible to add an option to trim the index from the image name? Slightly generalized, a "Trim-Name-Characters: <n>" would serve this use case and also allow people to trim an arbitrary count of leading characters from the image name.

Thanks,
Scott

keirf, 2021-08-01:
OLED width is >= 16 characters so DSKAxxxx shouldn't scroll anyway? You can also shorten `indexed-prefix=` in FF.CFG file.

sbelectronics, 2021-08-01:
It's also displaying the tag that I'm using for my img.cfg file. But, perhaps I could shorten both the tag and the prefix and make it all fit without scrolling. Thanks for the tip.

keirf, 2021-08-01:
You can get away with no name tag if you can instead discriminate by image file size. It's a possibility.

sbelectronics, 2021-08-01:
Thanks. All the images are the same size, so I will give that a shot!

BTW, Are you aware of any problems dealing with HFE images that have mixed sector counts? I'm using Epson QX-10 disks that apparently have 16 sectors of 256 bytes for the first 2 tracks, then 10 sectors of 512 bytes for the remaining tracks. Whenever I tried this using an HFE file, the computer would fail reads starting at track 2. When I converted them to raw format and manually described the geometry in img.cfg, it worked fine.

keirf, 2021-08-02:
No reason the HFE shouldn't work as it is simply a raw bitstream format which doesn't know anything about sectors. Could the HFE images be bad?

closed 2021-08-22 by keirf