keirf/flashfloppy #516: Write protection option in eject menu don't work

ariaththesage, 2021-08-01:
Hi!

I'm using Flash Floppy v3.21 and have my config set to respect the FAT read-only attribute instead of forcing it on.
I have enabled read-only in all my ADFs and they shows "Write protection" option in Eject menu correctly enabled

The problem comes when i decide to change the option from the Eject menu.

I have followed these steps:

1. Inserting an ADF.
2. Pushing encoder to enter Eject menu.
3. Navigating to "Write protection" option, that is correctly enabled.
4. Pushing the encoder to change "Write protection" from ON to OFF.
5. Navigating to "**Eject menu**" option.
6. Pushing encoder to go back to the ADF list.
7. Pushing the encoder again to enter the "Eject menu".
8. Navigating again to the "Write protection" option, that remains in OFF.
9. Repeating steps 5 and 6.
10. Selecting another ADF and inserting it.
11. Selecting the previously inserted ADF and inserting it again.
12. Entering the "Eject menu".
13. Navigating to the "Write protection" option.

The option is again ON, as it was at the beginning.

So, it seems to me that the read-only flag isn't saved correctly when changed from Flash Floppy menu...

Thanks for your assistance!

keirf, 2021-08-02:
You are correct it is a soft setting which does not get saved to the usb drive. I never thought about doing that. Perhaps it makes more sense.

gav-76, 2023-06-07:
Indeed, the current behavior might be confusing. I spent some time trying to understand why the write protection "disappeared", e.g. thought there's something wrong with my USB stick ;-) 
Reading this issue I then understood that's normal behavior.
 
Maybe, if you don't want to change read-only attributes on the USB drive, it could make more sense to call the option "temporary write protection", or something like that...

ariaththesage, 2023-06-07:
I case changing the ADFs attributes isn't possible or desirable, maybe we can create a file like "Game.adf.rw" as a workaround to keep track of the read-only status?

keirf, 2023-06-07:
Persisting the change in FAT attributes is I think doable. I wonder whether it is better than the current behaviour.