keirf/flashfloppy #517: Pin 34 (DSKCHG) stuck high

jsanpe, 2021-08-03:
Hi,

I am having an issue with my stm32-based gotek where pin 34 is stuck high. I've changed FF.CFG to debug the issue, always with the idea that this is a conf problem, but to keep it short:
- interface=ibmpc 
- pin32=auto | chg | nchg | low

All the 4 configs above result in pin 34 outputting 5v (4.8v to be precise). I know that FF.CFG is being read because I am changing the OLED fields to have visual proof that flashfloppy can read the file.

I also tried flashing the alt flashfloppy image with the IO test, and I can confirm that pin34 gets low as expected.

I am going to start reading the code to see if there is any hints as to why this is happening. I am running the unit disconnected from the PC for most of these tests just for convinience, but most of the promising configs I tried with my ibm ps/2 connected and always got the same error (Disk change line stuck high).

Thanks in advance for any tips you may have on this issue

keirf, 2021-08-04:
Pin 34 will only ever be low when the Gotek drive is selected. Are you measuring only while the drive is selected?

jsanpe, 2021-08-04:
Most tests I made with the gotek disconnected from the PC, so I am assuming the drive wasnt selected. Drive is identified as 0 or 1 with jumper S0/S1, is that correct? I think for the IBM PS/2 case S1 needs to be closed. I'll keep testing, I should at least see a low logical level with pin34=low and the drive correctly selected.

keirf, 2021-08-04:
Yes, jumpered as S1 the drive needs to be selected via input pin 12 pulled low for the drive to pull any outputs low.

jsanpe, 2021-08-05:
Used my oscilloscope to verify that pin 34 goes low indeed the moment that pin 12 is pulled low. The "Disk change stuck high" message still appears during boot, which probably indicates that my IBM PS/2 is expecting a bit of a different behaviour. I still cannot boot the reference disk from the gotek. I think my ibm planar expects a media sense supportive unit, which may be the issue. Is the whole media sense thing supported by flash floppy?

keirf, 2021-08-05:
I believe there are variations in PS2 interfacing for floppy drives. See the wiki Host Platforms page for some examples.

jsanpe, 2021-08-05:
We probably can close this issue, the stuck high message is clearly an issue with the expectations of the IBM PS/2 and not with Flash Floppy. I have an original drive, so I can try to identify the differences in pin34 between gotek and original if you are interested. For the time being, I will investigate on how to boot the reference disk. Thanks for all the support.

jsanpe, 2021-08-05:
A quick comment to share my findings for future reference. As it turns out, some IBM PS/2 models require a "media sense" compatible drive, which in a nutshell uses part of the odd-numbered pins as extra data lines. This makes the emulation of these drives unfeasible in the gotek, as all odd-numbered pins are tied to ground. 

We could modify the openflops project to add support for emulating these drives. The STM32 MCU used seems to have a bunch of non-utilized pins. In any case, the most relevant information about this matter I've been able to find is linked here: http://www.mcamafia.de/pdf/ps2_fdd_trm_s42g2194_00.pdf

Thanks!

jsanpe, 2022-07-23:
Hi again, I have recently been able to look into this issue I was having in the past. My IBM PS/2 was complaining about the Disk Change line being stuck high. I bought one of this cheap usb logic analyzers to understand what was different between the original IBM drive and the Gotek+FlashFloppy. Turned out there was no difference in the signals  !!

The interesting point is that I have been able to solve the issue by soldering a 500ohm resistor between pin 34 and ground. Reading IBM documentation it seems that the drive must be able to sink at least 4 mA when activating signals. Not really sure why the unit can't sink those through the mosfet. I am afraid I'm going to be having the same problem with the rest of the signals, not just DSKCHG.

With the information above, any of you have any idea of why, despite the analyzer perfectly showing low and high levels, the IBM floppy controller seems to be unable to detect low (active) signals?

keirf, 2022-07-24:
The Gotek output transistors are easily capable of sinking 4mA. Perhaps you have a bad transistor?

What is the voltage level at pin 34 when asserted, and connected to the PS/2? Is it significantly above 0V? What about other pins when they are asserted?

jsanpe, 2022-07-24:
Hi, after some more tests I realized that by using that Resistor I was basically driving the DSKCHG low all the time, so it wasn't a valid solution.

I've been taking some measurements with my scope
- The IBM floppy drive looks like this  (yellow=select, magenta=dskchg)
[image]
- The Gotek looks like this 
[image]
- Looks like the signal of the gotek is not exactly 0v when driven low, but around 0.6V. Not sure this could explain the issue though:
[image]

Back to square 1 :(

keirf, 2022-07-24:
0.6v should register as LOW. But it's still a surprisingly high voltage. Vce of the transistor at saturation should be maybe around 0.2v. What do other pins look like?

keirf, 2022-07-24:
3.92v isn't particularly high either for a signal pulled up to 5v. Should easily be enough to register HIGH though. 

Whats the 5v input power voltage reading as?

jsanpe, 2022-07-24:
I won't be able to take more measurements until tomorrow :( Family duties! Will send you some other voltages tomorrow. Any particular pin you would like me to probe?

Also, I'm a bit puzzled at the difference between the high level of SEL and DSKCHG for the Gotek. In the IBM floppy, both signals share a pretty similar high value, in the gotek SEL is 1V below DSKCHG. Could this be an idication of something?

keirf, 2022-07-24:
Pick any other output pin and compare high and low voltages. Also check the 5v line.

The Gotek design doesn't assign all inputs (such as select) to 5v tolerant lines. The result is that these lines get pulled down to around 3.3v or so via the input pin clamping diode. You will find some input pins (can't remember which) will read a higher voltage; they're the 5v tolerant ones. This is just a bad feature of the Gotek design. 

One thing I will say is the the new 435 based goteks use an mcu (at32f435) with more 5v tolerant pins and here all input pins are 5v tolerant. As they should be!

jsanpe, 2022-07-25:
Hi again, I have been reading the voltages and I'm getting good values across the pins I've tried (0.04V - 4.9V) including pin 34. Maybe there was some miscalibration going on with the scope yesterday.

I'm starting to think that the issue may be related to the expected impedance of the controller with respect to the floppy drive. If you look at the plots above, when the IBM drive is connected the SEL signal sits comortably at 4.8V or so. When connecting the gotek, the voltage drops below 4V. I measured resitance from SEL line to GND and 5V: IBM drive has 10k to gnd and 13k to 5V; Gotek 1.5K to gnd and 0.9k to 5V. Not sure this is meaningful, I can't really come up with an explanation of what's going on.

By the way, the IBM drive is a Sony Model MP-F40W-05. I haven't been able to find any documentation of this unit.

keirf, 2022-07-25:
Perhaps you need interface=shugart and pin34=auto ie perhaps pin 34 is RDY rather than DSKCHG?

keirf, 2022-07-25:
Other things to try are variations on chgrst= option in FF.CFG

jsanpe, 2022-07-25:
According to IBM documentation, pin 34 is DSKCHG. And comparing the signals in the logic analyzer, the sony drive and the gotek behave similarly. I'll give the chgrst options a try just in case. I'm still not sure any line will be read properly even if I get pass the DSKCHG stuck high. There seems to be some issue with the controller and it's expectations wrt to the drive. 

Thanks for all the help, I'll keep investigating the matter and report back in this ticket. Maybe it's a bit offtopic, so if you prefer me to close the issue just let me know.

jsanpe, 2022-07-25:
Ok, I think I've found the reason of this issue. I run an experiment in which I connected the SEL line directly to the DSKCHG line. Sounds like not the best idea, I know. But what I got in return was the stuck high message completely gone! This is how it looks in the scope:
[image]

So, if I disconnect this shortcut between SEL and DSKCHG, this is what I get in the scope:
[image]

In addition to the high level voltage differences, what can be observed is a 450ns+ delay in DSKCHG in responding to SEL getting active. Could this delay be the cause of the controller complaining about the line being stuck high? Just for reference, the IBM drive responds in only 30ns. Let me know your thoughts.

keirf, 2022-07-25:
Yes this is possibly a problem. Perhaps connect the gotek select jumper to ground so the gotek is permanently selected? There is a position on the jumper block that will do this. You need a select pin facing the gotek mcu and a ground pin. You don't want to ground the jumper pin facing the floppy header.

jsanpe, 2022-07-25:
That did the trick!!
I am now getting the issue about "The diskette drive is non-media sense", but that's to be expected. This PS/2 expects additional signals which can't really be implemented in the gotek, as most of the newly used pins are tied to ground (odd pins mostly). I have a modified version of the gotek to feed the expected values in these pins, I'll see where I can get with this.

Thanks for keeping up with this issue and helping out. I think we can close this now.

Best regards!

closed 2022-07-25 by jsanpe