keirf/flashfloppy #522: NEW GOTEK SFRC922AT3 it does not work on ACORN BBC MASTER COMPACT ?

proteus075, 2021-08-28:
I bought a new GOTEK SFRC922AT3 and installed the firmware (v3.24 v3.25 and v3.28) with decoder and oled display. But I can't get any (.adl .adf .adm .dsk etc.) image to work and despite having followed the CC.CFG settings suggested for acorn :

interface = shugart
host = acorn
index-suppression = no

the error message always appears while loading : "DISC ERROR 50 at :0/000002" 

the device is working on S0, and I can make a * CAT listing

I noticed is that the * VERIFY command when it indicates me on the BBC MASTER COMPACT at 50, on the gotek it is already at track 80 !

have you also encountered this problem? is there any way to fix?

see link to video -> https://youtu.be/Tw9SFFcIum8

keirf, 2021-08-28:
That seems strange. It shouldn't be stepping over track 80 I wouldn't think. Perhaps the cable or Gotek is bad? Do you ever see the track number go down, or always upwards?

proteus075, 2021-08-28:
the wiring is new and works correctly with the previous Gotek SFRC922C with
v3.22, I confirm what you ask me, the virtual head only reads one face of
the floppy, it does not go back and everything remains blocked, I have to
turn off and on to continue with the loading of a floppy .adl about 7/8
times, that's not normal ...

keirf, 2021-08-28:
Perhaps the Gotek is bad then. Do you have anything to cross-test it against?

proteus075, 2021-08-28:
I have several oscilloscopes but I can directly try another one, I update
you ...

keirf, 2021-08-28:
I mean can you test the Gotek against another retro system, or with a Greaseweazle/SupercardPro/Kryoflux, etc.

proteus075, 2021-08-28:
in an hour I'll create a .hfe with Greaseweazle and attach the screenshot,
see you later, thanks ...

proteus075, 2021-08-28:
I am shocked, the greaseweazle read both sides perfectly and the captured
.hfe / .adl file is fully functional ...

I tried to replace the cable without success, the error remains, the cable
works perfectly with the old gotek and with a SONY MPF-920 player

I'm running out of ideas ...

see foto ->

keirf, 2021-08-28:
Actually that may not test whether the Gotek can be stepped to lower track numbers. Try these commands:
```
gw seek 80
gw reset
gw seek 0
```
This ought to confirm whether the Gotek can be stepped from track 80 back to track 0.

proteus075, 2021-08-28:
I took the test:

I can go from 0 to 80 and back but if I send the gw reset command I can't
go back...

keirf, 2021-08-28:
Without the gw reset you're not really going back. Greaseweazle rememebes the track number and just counts 80 "backwards" but doesn't actually check you really went backwards! The gw reset means GW must resync to track 0 and will thus notice the problem. So.... it is a cable problem or a Gotek problem. On the Gotek check soldering at the floppy connector!

proteus075, 2021-08-28:
pin 18 is verified, pcb looks well assembled, do you have any other
suspicious pins i can check?

keirf, 2021-08-28:
Check pin 18 is 1k ohm resistance to 5v. Also check pin 18 continuity to the MCU (you can brush across the MCU pins and check one MCU pin buzzes out?). The physical pin number depends what MCU your board has.

proteus075, 2021-08-28:
*PROBLEM SOLVED*
just delivered me another new gotek SFRC922AT3 with CHIP AT: AT32F415CBT7
and it works perfectly ! I tried to compare the signals of PIN 18 with the
oscilloscope but there is a lot of jitter and I don't understand
differences, everything remained the same, cables etc. only the GOTEK has
changed and now the last tested works. It remains a mystery why the
previous GOTEK does not work, maybe the chip is damaged or some soldering
that I don't realize. I was unlucky to find a faulty GOTEK ! Thanks for
your cooperation and good luck, look at the photo ->

https://imagizer.imageshack.com/img922/9253/OzdP0d.jpg

*CASE CLOSED*

keirf, 2021-08-28:
That's the way it goes. Goteks are cheap and you got the less than 1 in 100.

closed 2021-08-28 by keirf