keirf/flashfloppy #523: Prepare my USP for Gotek immulator

biomedicalahmed, 2021-08-28:
Dear Sir;
 I perform Flash Floppy emulator as per your steps.
I have ( F-F ) in 7-segment of my emulator
I use ipcas program to make my USB have 0-100 diskette.
I put my system SW in four diskettes in one USB from 100 parts
When I just put its give me error E.3.4
I really see most YouTube video and thanks for all but clearly prepare your USB is not enough
I am trying to use emulator for Hitachi medical equipment (Elecsys 2010)
machine run by one old floppy disk drive
in order to update I am using another 4 diskettes.

your support are highly appreciated

keirf, 2021-08-29:
FlashFloppy does not use the factory supplied setup software. Instead place IMG image files on your USB drive. Call them 0.img 1.img and so on and they will be detected by the Gotek and selectable as images 000 001 and so on.

biomedicalahmed, 2021-08-29:
Thanks a lot for your support
I make two image 00.img and 01.img
it read and can see 000 and 001 when press button
kindly 
I attach my emulator to PC but not open floppy drive
this is normal?
second if I put my machine four images [00.img 01.img 02.img 03.img 04.img]
my machine will read this images without extracting them or I should but any empty image with my data disk of machine (data files + 00.img)

biomedicalahmed, 2021-08-29:
briefly my machine has system diskette(4) and data disk(1). I convert all these diskette source to image and rename them to (00-05)
machine should always run from files in data disk
could machine read files from image file

keirf, 2021-08-29:
> briefly my machine has system diskette(4) and data disk(1). I convert all these diskette source to image and rename them to (00-05)
> machine should always run from files in data disk
> could machine read files from image file

Yes that should work. Create an IMG file from each of your physical disks.

closed 2021-08-29 by keirf