keirf/flashfloppy #527: Basic version of FF needed

tootipn, 2021-09-02:
FF is really a great stuff! But I'm not crazy about those fancy functions like oled, rotary... as the original gotek model is more than enough in most applications. You may come up with a very basic version to suit the majority of the users. I suppose it could easily fit into the 48-pin MCU，even stm32f103(20k sram), or spi memory could be considered.

keirf, 2021-09-02:
The firmware already runs on such basic Goteks. So why would a separate basic version be needed?

tootipn, 2021-09-03:
> The firmware already runs on such basic Goteks. So why would a separate basic version be needed?

I'm refering to the coding. by taking the extra function away and focusing on your powerful img.cfg(the file can be enriched by contributors everywhere), I think the program size could be thinnered a lot.

keirf, 2021-09-03:
It already fits in 128k Flash and there doesn't seem any good reason to target a part with less Flash than that? Ideally if targetting a new MCU we'd want 64k+ RAM.

keirf, 2021-09-10:
I note this may happen if Goteks start shipping with less Flash (forthcoming AT32F425 part has 64k Flash only, and 20k SRAM). So perhaps something like this will happen then. But there is no work of this sort to do right now.

closed 2021-09-10 by keirf