keirf/flashfloppy #530: Rotary connection re-routing

andr229k, 2021-09-05:
I ordered a used AT32Gotek LQFP 64 on eBay, where the seller had a failed attempt to solder a wire into the pin-holes for Rotary-mod beside the OLED, maybe due to too much heat. 

Would it or is it possible to choose the back pins for rotary [Have the option to choose the routing], like we had at the start of the AT32 migration, so I can save this Gotek?

keirf, 2021-09-10:
It is not worth the effort for one Gotek.

closed 2021-09-10 by keirf

andr229k, 2021-09-16:
Issue resolved by using alternative entry point as per attached screenshot
[image]

keirf, 2021-09-17:
Good call. Hadn't realised it was this model with the extra pins!