keirf/flashfloppy #531: An unseen model: SFRA619

denjhang, 2021-09-06:
[image]
[image]

keirf, 2021-09-06:
Does it work?

denjhang, 2021-09-06:
I tried it a bit, and successfully uploaded the firmware using DFU mode (I used a USB male-to-male cable and directly connected it to my computer). The OLED mini display has been installed, but I found that the modified version of the PCB does not have DT And CLK leads, so it is impossible to install a rotary encoder (unless I use a flying lead).

keirf, 2021-09-28:
I think it must be an old model.

duca750, 2021-09-28:
It is not old, I have received two for my equipment today, has someone found a solution for the rotary encoder? Thanks.

keirf, 2021-09-28:
Depending where you got it from it could have sat in stock a long time. It doesn't look very shiny and new. Compare usb socket on this ticket of a brand new model: https://github.com/keirf/FlashFloppy/issues/539

closed 2021-10-04 by keirf

wistal, 2023-12-03:
Just leave a comment to confirm this model works, I got this model recently, after flash the FF firmware, it works well, and it also supports OLED screen.
UART header in J4 and 4 pin i2c header in J7 as above picture.

WellsWang, 2025-04-19:
> It is not old, I have received two for my equipment today, has someone found a solution for the rotary encoder? Thanks.

You can fly wires from PC10/PC11 (PIN 78/79 ) of STM32F105 to enable rotary encoder. I tried it and it works fine.