keirf/flashfloppy #532: New GOTEK with AT32F415RBT7

Giants92, 2021-09-09:
[Moved to a new issue]

Hi !
I use on my all Gotek HXC and FF firmware.
Recently I order 2 Goteks with AT32F415CBT7
First, I try to use it with FF last version (so 3.28 and tested also beta 4)
No problem to flash, Oled work, navigation work, ...
BUT, interface NOT detected and so, don't work, on Amiga (500, 600 no matter)
NOT detected with or without FF.cfg
I think it's 'normal' and because the last firmware 3.28 is not adapted to this new Chipset ?
If it's the case, any idea when it will be supported ?
Thks a lot.

Already tested, Gotek not detected on my computer.
...

I'v just re-try again.
- re-flash with 3.28 Firmware my USB-Dongle (holding both buttons) : Ok, DONE ! 
- Format this dongle on Fat32 and put a adf image Test_kit_harware.adf (your tool)
This ADF already tested and working on other Gotek and under winUae, no problem.
- Put this dongle on Gotek, Boot Amiga and.... Nothing, interface not detected.
Seams working but not.

On boot : 
- FlashFloppy 3.28 appear on OLED
- Adf Image name appear on OLED but stick on 000/000 T:00.0
- No activity. (no buzer activity)

Info : PCB SFRKC30AT4 with AT32F415RBT7
This Amiga (A600) work good (with other Gotek (SFRC922D under 3.28 for example) and Original Floppy reader.
On This new gotek PCBs, No jumper only S0 connected. (and buzzer)
Tested on another gotek AT32F415RBT7, same problem... seams work good but, interface not detected.

[image]
[image]

Tested also on other Amiga (500 Rev6)
Same... interface not detected.

If I remember well, when the ribbon is wrong connected, a message appear on OLED
In my cas, no matter the connexion is good or bad (so ground reversed or well connected)
No message to warn.
Strange...
I also try the 'Motor Signal' hack, always same problem.
I try also With or without ff.cfg, no matter, same issue.

keirf, 2021-09-09:
I have report from Facebook group that someone has one of these working. The board looks identical except the model name is printed as SFRKC30.AT4 instead of SFRKC30AT4.

Giants92, 2021-09-09:
Yep, strange.
If you wan I can do some test.
I have logical analyser and osciloscope.

keirf, 2021-09-09:
I have further report from maker than the two models (with and without dot) are identical just from different PCB factories and hence he added the dot to discriminate them.

And someone else has reported the SFRKC30.AT4 variant works for them.

You may want to buzz out floppy interface pins to MCU pins. It's a little tedious. You'd do it with reference to the MCU pinning in the datasheet http://www.arterytek.com/download/DS_AT32F415_V1.03_EN.pdf

And check that floppy pins are connected to MCU port pins (PAx, PBx, etc) same as in the original Gotek schematic: eg https://cortexamigafloppydrive.files.wordpress.com/2021/01/gotek_usb_floppy_emulator_schematic_resized1.jpg

keirf, 2021-09-09:
Beyond this, I need one of these offending PCBs in my hand to be able to test it myself.

keirf, 2021-09-09:
I have someone sending me a SFRKC30.AT4. Note it's the dotted model, but is expected to be identical to your PCB.

keirf, 2021-09-09:
By the way, actually this RBT7 chip is pin compatible with the original STM32 chip. So you can simply buzz to physical pin numbers on the MCU package and cross-check those to the STM32 schematic. There's not even any need to map to PAx/PBx/PCx port numbers.

Giants92, 2021-09-09:
Work in Progress.
According to the PDF : LQFP64 we have
[image]

Can you validate that please.

keirf, 2021-09-09:
Correct

Giants92, 2021-09-09:
Ok.
So, et voila :)

[Gotek_AT32F415RBT7.pdf](https://github.com/keirf/FlashFloppy/files/7136878/Gotek_AT32F415RBT7.pdf)

Giants92, 2021-09-09:
I trace each pin, one by one.
for 'Signal' on this file, I took back the information under your link of DS_AT32F415_V1.03_EN.pdf LQFP64

Giants92, 2021-09-09:
Hummm, some mistake of my side, I just re-check that.

Giants92, 2021-09-09:
Well.
For me, all is good.
Except PF6 & PF7 not connected to ground but I don't think it's the problem here.

[Gotek_AT32F415RBT7.pdf](https://github.com/keirf/FlashFloppy/files/7137873/Gotek_AT32F415RBT7.pdf)

Well, you'r right, It should work...
Maybe the Ready signal ?
I going to see signals with my osciloscope.

Giants92, 2021-09-09:
Just add some information.
I trying with Kickstart 3.1, Enter to Early-Boot and see if Amiga See DF0
It saw it.
On Amiga Power I can a floppy access (very quicly and short from y buzzer.... click...)
L5 LED Stuck powerOn.
Enter to menu et select 'Exit and re-insert' but... no change. (and not sound from my buzzer)

Check of Dsk_Change (pin2) is well connected to U7 pin4.
Check od Dsk_Change directly connected to Gotek floppy connector --> My osciloscope.
PowerOn Amiga -> Dsk_Change go to 5V and stuck to it
Try to Enter in the menu et select 'Exit and re-insert', validate it... and...
No change on my scope, Stuck to 5V.

Giants92, 2021-09-09:
Here some capture.

Hardware : Amiga600 + FlashFloppy Firmware 3.28 + Dongle USB with Only 1 adf file (test kit)

With Gotek SFRC922D - STM32F105
Graphic start with PowerOn 5V and boot quickly to the adf file
[image]

Same USB dongle but on the New Gotek SFRKC30AT4 - AT32F415RBT7
[image]

keirf, 2021-09-10:
Have you tried another usb drive? Maybe the Artery chip is fussier.

Giants92, 2021-09-10:
Another usb drive ? For ? My Logic analyser ?
No need to, I have prove also with my Osciloscope (hantek), no signal present on this news Gotek (except Motor_On)

This 2 new gotek have been programmed with 'Artery ISP Programmer_V1.5.47'
First Broke the read protection and erase the chip and second program with the last FF firmware.
Same process like with STMFlashLoader.
On the first 'new' gotek, leave like this and test It (So don't respond like floppy, but well progammed (checked after program it) and I See on the OLED the menu of FlashFloppy.
On the second 'new' gotek, same programmation but after I change to FF v4 (beta) with USB dongle and press 2 buttons when Amiga Boot.
So, on this 2 new goteks, tested on Amiga500 and Amiga600
Tested with FF 3.28 and beta 4, they behave the same.
Seams working (OLED see 3.28 on boot, and I can move into menus)
It detected on Amiga (early boot see DF0 and if I made hot swap, I can check with menu of 'Amiga Test Kit')
But.... stuck in coma.

If I understand the 'step' of floppy reader (and so gotek)
[image]

Amiga probe if they are S0 or S1 present into the chain.
If yes, Amiga active 'Motor_On' to have some rotation (2 I think, seeking index maybe ?)
Check Ready_Signal, and send go to 'T00'
Not sure about the order of process...If you have a link !

Maybe the Ready Signal is not well replied by this gotek and so, process is stopping.
Maybe is an electronic problem.
Maybe I have 2 goteks broken.
I don't know but I'm still searching :)

keirf, 2021-09-10:
Could be a drive-select issue: That is floppy pin 10 (SEL0) needing to reach MCU pin 14 (PA0).

If the MCU does not think it is selected, it will not drive any output pins at all.

Trying another USB drive in the Gotek is just belt and braces. It's an easy test. What is your current USB drive make model?

keirf, 2021-09-10:
Just received SFRKC30.AT4 model in the post. Jumpered to S0, tested on my Amiga 500... It works of course ;)

Giants92, 2021-09-10:
#Could be a drive-select issue: That is floppy pin 10 (SEL0) needing to reach MCU pin 14 (PA0).
But, In this case, I will not have DF0 detected on the Early-Boot isn't it ?

When you tell 'USB drive' you spoke about USB stick ?
It's a 32G, I don't know the name of this model but it's a 'standard', it's work good on other Gotek.
It's detected by this new gotek and I can choose/naviguate on it if I write a lot of ADF file on it.
I use it on this new gotek to update/write the firmware 4Beta to 3.28, no problem.
I can test another 'usb drive' no problem.

#Just received SFRKC30.AT4 model in the post. Jumpered to S0, tested on my Amiga 500... It works of course ;)
Sh... , what is wrong on my side... very strange.

I going to test another 'usb drive'
Proble/trace signal(s)
Re-flash with anoter tools...

Another strange thing.
So I flash this 2 new model with USB A-A cable (same cable I use since a long time, it's work good)
But now, On this 2 new models flashing in 3.28 FF, If I try to re-flash with A-A cable and so wire on Boot-3V and Tx-5V.)
...Device NOT detected on PC... nothing...
Very very very strange.

Giants92, 2021-09-10:
Humm I think I find It. (need to correct to validate that)
They are line cut betwen M0-S1-S0
So indeed, on one of the goteks, SEL0 is cut between AMIGA and MCU
Just 2mn to correct that and check on another gotek if it's the same problem

Giants92, 2021-09-10:
Goal. :)
That was it.
Line of SEL0 was cutting, with a wrap, it's work :)
Time to check on the second gotek if it's the same problem.
I traced the SEL0 signal to VIA to MCU, NOT directly pin10 to MCU
There was a cut between the VIA (coming from MCU pin 14 (PA0)) and the jumper(s)
It's normal in this cas amiga was detected anyway DF0 on Early boot ?

keirf, 2021-09-10:
Amiga always assumes it has DF0. Since all Amigas (except CDTV/CD32 "consoles") have at least one floppy drive.

closed 2021-09-10 by keirf