keirf/flashfloppy #537: OLED acts un-initialized at boot, AT32F415CBT7

cshunt, 2021-09-25:
Just received my gotek drive and added flashfloppy. Works great with the 7 segment display. I added a .91 OLED display and tested it out. First time I booted, the oled showed static. Inserted the usb stick and it seems to work fine... put it all back together and all i can get when turning it on is static on the oled display. Everything seems to be working in the background, disk selection and drive works, just not displaying. If i unplug the oled and plug it back in, the screen turns on and works as expected. 

I tried adjusting the ff.cfg to add i2c delay up to 15000 ms, doesn't seem to make a difference. Specified the oled as well.

It appears to be interference to i2c on the power supply lines and/or fdd cable.

My solution at the moment is a 1000uF capacitor on the vcc line of the oled charged by a 220 ohm resistor... If I delay the oled startup it seems to display correctly.

One thought was perhaps with the artery chip, a possible wait to poll i2c devices until power has settled down on these old systems. Or this could just be a fluke with the components i have, if its the only time this happens.... 

Any thoughts are appreciated. 
[image]

keirf, 2021-09-26:
Probably the specific OLED display. They're cheap parts with poor QC. You really need to try another.

keirf, 2021-09-26:
The RC delay will be delaying FlashFloppy startup too as it shares 3v3 with the OLED. If it works it works but a neater solution for this display might be to identify the RC delay on its reset line and increase R.

cshunt, 2021-09-27:
I suspect the same, cheap parts (<$2). I have the RC tied only the oled so flash floppy is booting up like normal. (I tried increasing the capacitor on C17, actually I added the capacitor there, C17 was empty.) I thought the 3.3 might be unstable but that didn't make a difference. I did notice that flash floppy boots up without the power connector, must be enough draw or feedback from the floppy cable. 

I don't have another oled to test with so I'll close the issue. 

Btw, flashfloppy works great. Thanks.

closed 2021-09-27 by cshunt