keirf/flashfloppy #539: Board SFRKC30.AT2 with AT32F415KBU7-4 MCU

doragasu, 2021-09-27:
Today I received this board, and flashed latest 3.28 firmware. Unfortunately, it doesn't seem to boot (or at least the display shows nothing at all). I opened the board and to my surprise it has an AT32F415KBU7-4 MCU in a very small QFN32 4x4 mm. I searched the datasheet and it seems it has 128 KiB Flash and 32 KiB RAM, so I suppose ti could be supported.

This is how the board looks:

[image]

Has this board been spotted elsewhere? Will it get support?

doragasu, 2021-09-27:
Bottom of the board:

[image]

keirf, 2021-09-27:
Please try the following test firmware (you need to be logged in to Github to download it): https://github.com/keirf/FlashFloppy/suites/3840876534/artifacts/95197708

RetroClinic, 2021-09-28:
Hmm, I got a batch of these today as well. I'll try a build on a few in the next few days, and try whatever firmware you've got at the time, and report back.

keirf, 2021-09-28:
The firmware just needs test feedback. Then I will do a release.

keirf, 2021-09-28:
It would also be handy to have a vanilla factory unit myself. Could you send me one?

RetroClinic, 2021-09-28:
I’ve sent you a FB PM.

From: Keir Fraser 
Sent: Tuesday, September 28, 2021 6:32 PM
To: keirf/FlashFloppy 
Cc: RetroClinic ; Comment 
Subject: Re: [keirf/FlashFloppy] Board SFRKC30.AT2 with AT32F415KBU7-4 MCU (#539)

It would also be handy to have a vanilla factory unit myself. Could you send me one?

doragasu, 2021-09-28:
> Please try the following test firmware (you need to be logged in to Github to download it): https://github.com/keirf/FlashFloppy/suites/3840876534/artifacts/95197708

Thanks for the quick reply! I flashed the "debug" firmware (I do not know if I had to flash this or the "non-debug" one) and it seemed to boot (I saw the F-F in the 7 segment digits). But then I tried plugging a USB stick, powering it again and it has stopped working, It does not seem to boot anymore (the digits are blank again). I have tried reflashing, but although the flash process completes successfully, it does not seem to boot anymore.

Should I try the non-debug firmware or any of the "alt" images?

BTW I flash from Linux using `dfu-util`. I suppose this does not matter, but just in case I am using this command:

```bash
$ dfu-util -a0 -D FF_Gotek-016e032-debug.dfu
```

It worked flawlessly for my other Gotek in the past.

doragasu, 2021-09-28:
Oh, I discovered what happened, my bad, the problem was my USB stick has a FF.CFG configuring an OLED screen, and that made the 7 segment display to stop working. I plugged a 128x64 OLED and it now boots and read my files properly!

Now I have to solder a 26 to 34 pin adapter and attach it to my Amstrad CPC 6128 to see if it works, but it looks promising!

keirf, 2021-09-28:
Please use the non-debug firmware for testing.

Knuckles112, 2021-09-30:
> It would also be handy to have a vanilla factory unit myself. Could you send me one?

Hi keirf,

I also have received some of these new boards (model board SFRKC30.AT2). Testing now with my retro system... If you still need it, I can send you one of these today by urgent post :-)

keirf, 2021-09-30:
If you need rotary encoder pinning, I can explain it here. I don't have a diagram currently for the new 6p header

RetroClinic, 2021-09-30:
I'm just assembling some of these now alongside older ones. I've put pins into all the header spaces just incase, so knowing where the encoder goes would be useful, if it's different from the older board that you had to connect it at the back near the power socket.

keirf, 2021-09-30:
Left to right, top to bottom, starting with pin JK: Select, unused, rotary DT, rotary CLK, Ground, 5v

mnoga23, 2021-09-30:
I have one of these new Goteks with SFRKC30.AT2 board, but how can you flash that 256kiB .hex file onto it if its flash is just 128kiB?

```
Interface serial_posix: 57600 8E1
Version      : 0x32
Option 1     : 0x46
Option 2     : 0x01
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : Up to 20KiB  (512b reserved by bootloader)
- Flash      : Up to 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
```

RetroClinic, 2021-09-30:
Because HEX is in ASCII, it gets converted to actual bytes by the software.

From: mnoga23 
Sent: Thursday, September 30, 2021 3:42 PM
To: keirf/FlashFloppy 
Cc: RetroClinic ; Comment 
Subject: Re: [keirf/FlashFloppy] Board SFRKC30.AT2 with AT32F415KBU7-4 MCU (#539)

I have one of these new Goteks with SFRKC30.AT2 board, but how can you flash that 256kiB .hex file onto it if its flash is just 128kiB?

Interface serial_posix: 57600 8E1
Version      : 0x32
Option 1     : 0x46
Option 2     : 0x01
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : Up to 20KiB  (512b reserved by bootloader)
- Flash      : Up to 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB

mnoga23, 2021-09-30:
Ok it makes sense, I will try that firmware as soon as I get my USB to TTL interface working and take the backup.
What am I doing wrong if firmware backup comes 128kiB of 0xFF ?
I tried swapping Tx and Rx around but then it just shows 'Failed to init device.' so the cables must be correct way.

keirf, 2021-09-30:
You can't back up factory firmware. When you turn off read protection the Flash is mass erased.

mnoga23, 2021-09-30:
ok then, in that case, since i bricked it already I flashed your firmware onto the device. it shows F-F when powered on. 
I formatted usb stick to FAT32 and put some .IMG files onto it.
FF.CFG is in root and has just these 2 lines:
```
interface=ibmpc
write-protect=no
```

It seems to be working. I can go up/down with keys, and images are properly opened on a PC.

Anything else I can try to help?

keirf, 2021-09-30:
No apart from testing OLED and rotary that's it. If you're not fitting those, you're done. Thanks!

RetroClinic, 2021-09-30:
Ok, done a test with all the mods. Seems to be all ok. I can only test BBC functionality but works for read and write in DFS and ADFS modes, with both SSD, ADF, and HFE.

The only thing you've either got wrong, or I read it wrong, is the connection for the rotary CLK and DIR. If you look at doragasu's photo at the top, you've got the block of three:

1 - 2 - 3
4 - 5 - 6

Where 4 is marked JK and 5 is marked J8. The CLK and DIR go onto pins 2 and 5. On the ones I've done, I've still wired the select into the usual place at the back as normal, and that works fine. Will that still be supported or should I move that as well to the 6 pin connector?

keirf, 2021-09-30:
I think I'm right I was counting rotated 90 degrees clockwise compared with your diagram. So we are actually equivalent. This is why I need a diagram!

keirf, 2021-09-30:
JA will continue to be supported for the push button but I would have thought connecting all rotary pins at the new header would be neater.

RetroClinic, 2021-09-30:
Yes, and in future builds I'll do that for these newer boards. I just wanted to make sure for the ones I've already done, that the functionality won't drop off if the end user updates the firmware.

Great support for these, keep up the good work! Has the new unit arrived with you yet?

keirf, 2021-09-30:
Yes it has, thank you very much!

doragasu, 2021-10-01:
Finally got time to solder the adapter board, plug the gotek to my CPC6128, add a tiny speaker and the accept button. Everything works flawlessly, thanks!

[image]
[image]

closed 2021-10-01 by doragasu

keirf, 2021-10-02:
This board is now supported in latest stable release v3.29.

Also in latest development (testing only) release v4.1a.

doragasu, 2021-10-03:
Awesome, thanks!

keirf, 2021-10-10:
There is a visualisation in the wiki

brianrudie2, 2022-03-25:
This board is really problematic, I have set this up with an Akai S3000XL Sampler and it reads ok, the rotary, OLED, and eject work well, but every time I attempt to write to floppy image from it, it stops reporting bad floppy disk track. Has anyone else experienced issues with this? I have tried multiple blanks that I know have worked well in the past but no luck.

keirf, 2022-03-25:
Have you tried v4.4a?

brianrudie2, 2022-03-25:
> Have you tried v4.4a?

No , 3.30 , where can I get that one?

jamilxushvakov, 2022-05-05:
Эта плата действительно проблематична, я установил ее с помощью семплера Ensoniq TS-10,и она читается нормально, поворотный экран, OLED и извлечение работают хорошо, но каждый раз, когда я пытаюсь записать образ дискеты с нее, он перестает сообщать о плохой дискете или "disk not formatted".  Я пробовал v4.4a v3.29, v3.30

jamilxushvakov, 2022-05-05:
уважаемые кейрф! какой пошивка подходить лучшей

jamilxushvakov, 2022-05-05:
[image]
если так делаю она будет работать

jamilxushvakov, 2022-05-05:
через инвертор

bangstk, 2022-07-04:
Hello,
I have SFRKC30.AT2, but it DOES have a JC marked jumper, and a corresponding board trace for it. Perhaps a stealth new revision?

[image]

Incidentally I have some IO issues (WGATE never reads as HIGH in the IO test firmware, and TRK00 always outputs 5V despite the output toggling)

Is it possible something in the firmware is detecting the board wrong due to the restoration of JC?

keirf, 2022-07-04:
The io test firmware never got updated for the new Gotek boards. 

Does the board work in normal operation?

Does the JC jumper work can you tell?

bangstk, 2022-07-04:
No, I am having trouble using it with a Yamaha V50 (the synth says "Bad disk!" after any read/write operation)

I decided to discuss in a new issue #663