keirf/flashfloppy #540: Oled scrolling issue on 3.28

Phlavio-SPR, 2021-10-04:
Not sure if I'm missing something, but I can't make large texts stop scrool at the end of text.
"display-scroll-rate" and "display-scroll-pause" doesn't seen to change anything on oled behavior.
When I stop navigating and the showed directory name is larger than display width, then the name keep scrooling endless.
My current settings:

display-type = oled-128x64
oled-font = 8x16
oled-contrast = 143
display-order = 1,3,0d
display-off-secs = 240
display-on-activity = yes
display-scroll-rate = 200
display-scroll-pause = 2000
nav-scroll-rate = 80
nav-scroll-pause = 2000

Thanks!

keirf, 2021-10-04:
You can't turn it off. If you are in navigation mode (which you must be if it's a folder name that is scrolling) then you can increase nav-scroll-pause up to 65535. Also  display-off-secs will eventually stop scrolling by dint of blanking the display!

Phlavio-SPR, 2021-10-04:
Thanks for your prompt reply.
I see. So it's not a problem, but a feature.
If I understand well:
"display-scroll-pause" only affects filenames when already selected. It adds a pause at the beginning AND at the end of the text.
"nav-scroll-pause" affects file and folder names, but adds a pause only at the beginning of the text, not at the end, so the name will scroll infinitely.
They act differently, which seemed odd at first, but it's actually nice to differentiate the navigation text from the text of an already selected file.

keirf, 2021-10-04:
If you want selected filename to endlessly scroll, you can set display-scroll-pause to 0.

closed 2021-10-04 by keirf