keirf/flashfloppy #548: Artery-based gotek brokes ADF's and HFE's with Amiga disk images

nihirash, 2021-10-11:
I've checked it with SFRKC30.AT4.7 board, 3.29 FW.

[image]

It happens not every time but when I try store something(and not always).

If it's important - I'm using OLED+Rotary encoder mods. 

Amiga is A500 with Kickstart 1.2. 

Corruption of image happens not every time. It's look like random thing. 

With real floppy drive there are no corruption of disks(but I really have only 4 DD disks at home now, couple of them is kickstart disks). 

Issue happens with ADF and/or HFE images.  I've tried 3 different flash drives and around 5 SD2USB converters with SD cards with the same result.

nihirash, 2022-01-20:
As I remember 4.2a had maximum stability. 

Sorry, my father dead from covid and I haven't access to my gotek - I'm in other city now. 
I'll back to my hometown at February and can check it again.

keirf, 2022-01-20:
Condolences for the loss of your father.

And no hurry on this, I was just bug scrubbing.

keirf, 2022-06-22:
A follow up on this: New AT32F435 Goteks will be available in about a week's time. If you're still watching this issue, might be worth obtaining one and testing with that. The new chip has 384kB RAM, way more even than the old STM32 chips.