keirf/flashfloppy #561: Screen does not work now, but before it did go

Dunkler2Engel, 2021-10-29:
Hi i have a Gotek with AT32F415CBT7.
The Screen it works before perfect, but now, not do nothing. I was try with Oled 0,96 and with the numeral screen that gotek has by default.

But in both cases dont show nothing, i has try update the gotek firm with 4.x but the problem stills.
The gotek works fine, and the rotary too y can change disk and works, but i dont know what im has selected ^ ^.

Anyone with the same problem? is there any other points to sda & scl in the board? 
The pins of the screen has voltage but in any case, can i use the 3,3 and gnd of psu?

But i think the chip is dead in this pins :(

Dunkler2Engel, 2021-11-05:
I do more test, and the screeen pins are dead, i was try with 2 more screens, and with 3-5v & gnd from psu pins, the screen has power but it still in black.
The SDA and SCL has continuity with the Artery chip, in the pins 3 and 4 in this side but as said, not show nothing the screen :(.
I think i need to buy another gotek, Without screen is not the same

keirf, 2021-11-05:
Try a full config reset (press rotary button for a few seconds *after* power on with *no* stick inserted).

Dunkler2Engel, 2021-11-05:
Thx, for the tip, but not work , if press after power on the led blink and this is all, if i reboot the gotek it have black screen. If i try to press before power on, amiga logo take a few more seconds to load, but the screen still in black. And i made a firmware update with the same result. The flashFloppy is good, i think my problem is with the chip.
But it is funny, the only thing that does not work is the screen, I would prefer that rotary encoder is what fails xD.
In any case, thanks for your answer and for the FlashFloppy that I have been using for a couple of months and it is going great

keirf, 2021-11-05:
Okay yes sounds like a hardware problem. Quite unusual though!

closed 2021-11-05 by keirf