keirf/flashfloppy #562: Ensoniq ASR-X Pro and disk not readable / No loadable files error

andr229k, 2021-10-30:
**Enviroment:**
Ensoniq ASR-X Pro with internal firmware 3.04 and Gotek [SFRKC30AT4] firmware 3.29 together with a USB-stick formatted as MS-Dos [FAT32] on a Macbook.
´
Jumper is set on S0 and following settings in the ff.cfg:
interface = ibmpc-hdout
host = unspecified [This option due to following topic about ASR-10 -> https://github.com/keirf/FlashFloppy/issues/139

**Symptoms:**
When I try to load .HFE and .IMG files, then I experience error messages like “disk not readable” and “No loadable files”.

[image]

**Diagnose:**
•	The green led is blinking on the Gotek when executing a disk function, which means that it receives a handshake from the ASR-X Pro.
•	Internal firmware 3.04 is the latest firmware and is important to be able to load example ASR-10 and EPS 16 + disk etc.
•	ASR-X is Shugart, while ASR-X Pro is IBM PC, this is confirmed here [https://hxc2001.com/download/floppy_drive_emulator/support.htm]

**Actions:**
•	Testing other USB-sticks.
•	Testing ibmpc instead of ibmpc-hdout as Interface in ff.cfg
•	Testing unspecified instead of Ensoniq as platform in ff.cfg

Config attached, any suggestions?
[FF.CFG.txt](https://github.com/keirf/FlashFloppy/files/7446320/FF.CFG.txt)

keirf, 2021-10-30:
IMG files will require `host = ensoniq`. What exact byte size are your IMG files?

HD IMG files may be unreliable with Artery MCU on v3.x. You could try Flashfloppy v4.2a.

andr229k, 2021-10-30:
I will try changing the host back to Ensoniq, I have attached two images [.HFE and .IMG] and the ff.cfg, which im planning to test.

I normally prefer to stay with the latest stable confirmed version, but I will give 4.XX a shoot. 
[FF.CFG.txt](https://github.com/keirf/FlashFloppy/files/7446420/FF.CFG.txt)
[ASRX-FDX100-Omniflop.img.txt](https://github.com/keirf/FlashFloppy/files/7446421/ASRX-FDX100-Omniflop.img.txt)

[TECHSTART-30.zip.txt](https://github.com/keirf/FlashFloppy/files/7446422/TECHSTART-30.zip.txt)

andr229k, 2021-11-02:
> 
> 
> IMG files will require `host = ensoniq`. What exact byte size are your IMG files?
> 
> HD IMG files may be unreliable with Artery MCU on v3.x. You could try Flashfloppy v4.2a.

USB Floppy Emulator has been updated to 4.2.

Below is the errors, which the Ensoniq ASR-X Pro is responding to above images
the .img -> No Loadable Files!
the .hfe -> Disk not readable

[FF.cfg.txt](https://github.com/keirf/FlashFloppy/files/7461489/FF.cfg.txt)

keirf, 2021-11-02:
Did you try `host = ensoniq` and IMG files with v3.29 too?

closed 2022-02-19 by keirf