keirf/flashfloppy #563: Broken build with binutils 2.36+

ejona86, 2021-10-31:
This is mostly a brain dump; I've spent enough time on this for the moment. If/when I feel like it I can create a minimal reproduction and file a bug with binutils.

I'm on Arch Linux, so I tend to see things early. I saw 2.36 was broken a while back, but just downgraded to 2.35. I tried with 2.37 today and it is still broken, so my hope it would "fix itself" doesn't look likely to pan out.

Based on my investigation, this really seems like a bug in binutils `as`, although we may want to change something on our side as well.

Symptom:
```
/tmp/ccjS07gm.s: Assembler messages:
/tmp/ccjS07gm.s:27: Warning: dwarf line number information for .data.ramfuncs ignored
...(lots more of these)
/tmp/ccjS07gm.s:26995: Error: leb128 operand is an undefined symbol: .LVU16
/tmp/ccjS07gm.s:26996: Error: leb128 operand is an undefined symbol: .LVU16
/tmp/ccjS07gm.s:27017: Error: leb128 operand is an undefined symbol: .LVU60
/tmp/ccjS07gm.s:27018: Error: leb128 operand is an undefined symbol: .LVU60
/tmp/ccjS07gm.s:27039: Error: leb128 operand is an undefined symbol: .LVU60
/tmp/ccjS07gm.s:27040: Error: leb128 operand is an undefined symbol: .LVU60
```

The warning provides a strong clue to the sort of thing going on, but before f0edf11 it just said `.data` instead of `.data.ramfuncs` which made the message very unhelpful.

To see more info, I used `make CC="arm-none-eabi-gcc -save-temps"` and then looked at the lines in the temp file. The warnings are for lines like:
>     .loc 1 180 8 view .LVU16

File 1 is `gotek/floppy.c` and line 180 is an ordinary-looking line in `_IRQ_SELA_changed`. The error lines are label references deep in raw, incomprehensible debug information. [The `as` documentation](https://sourceware.org/binutils/docs/as/Loc.html#Loc) clearly says that the label would be created by the `.loc` op, which is why this seems clearly an `as` bug, but it's probably easy to pass the blame around.

The error mentions dwarf, and sure enough, removing `-g`, using `-g1`, or using `-gdwarf-2` fixes the error (but not necessarily the warning).

It seems this problem was introduced because of https://sourceware.org/bugzilla/show_bug.cgi?id=26850 . A later commit dropped the warning for DWARF 2, https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=be07043ea8ea9b22a8db1ff46b557321a7714e9a .

From the above, it may _seem_ obvious what the problem is: we have code in `.data`, but modern `.data` is typically not executable and so would make no sense to have code debugging details. Using a `.text` section instead of `.data` section for those special functions in `gotek/floppy.c` does fix compilation. But I will also note that in the temporary output we can see the segment is defined as executable:
```
    .section    .data.ramfuncs@,"ax",%progbits
```

It is probably possible to work around this by using `.text` and tweaking the linker configuration to put the functions in RAM. But meh.

keirf, 2021-10-31:
The reason for changing the name to `.data.ramfuncs` was another possible toolchain bug, also dwarf related, in which function text was getting padded with 0s instead of nops, and this was upsetting a later stage of the toolchain emitting dwarf info.

Also note that the segment is not explicitly defined as executable due to the truly disgusting use of `@` -- if we drop that we get other toolchain errors. But as is the parameters after the section name are in a comment section.

From my side, the executive summary is that I have not found a simple foolproof method to put functions into RAM. A simple requirement for developing firmware you would think!

fook42, 2021-11-05:
just to add a :+1: 
same problem here.. running Ubuntu 21.10 (impish) 
installed binutils-arm-none-eabi/impish,now 2.36.1-0ubuntu1+14build1 amd64

keirf, 2021-11-05:
Ok, I fired up Docker and tried Ubuntu 21.10 myself. The `@` comment character skankiness is to blame. If I remove that then the debug build completes with no errors. What I do get is a warning (but not an error) which must be why I put the `@` in the first place: A clear case of the cure being worse than the disease.
```
Warning: setting incorrect section attributes for .data.ramfunc
```

keirf, 2021-11-05:
Okay this whole thing was my bad kludge, sorry. Should be fixed now in `master`. If everyone is happy we can cherry-pick to `stable-v3` and close this ticket.

ejona86, 2021-11-06:
Ah, so removing `@` is the piece I was missing. The new commit looks straight-forward and works well on my machine. Seems good to backport.

closed 2021-11-06 by keirf