keirf/flashfloppy #564: Gotek in Tandy 1000 SX will not boot from IMG Boot images (DOS)

XGAMER777, 2021-11-04:
Gotek is working. The green light goes on when an image is selected, Image name and track are displayed on Oled display.
Tried using both S0 and S1, jumped JC also. Using a standard floppy cable. Tried multiple boot floppy images from different places. Tried creating an FF.CFG. Flashed to latest stable firmware. But now success.

keirf, 2021-11-05:
Is the host machine known working okay?

XGAMER777, 2021-11-05:
Yes. Works perfectly. It boots up from 5.25 360K floppy.
See attached.

Ruben Corbo

keirf, 2021-11-05:
Can't see attached as Github removes attachments from email replies. Can you check your 5v line voltage.

XGAMER777, 2021-11-05:
Oh!? Did think that power could be an issue.
It's running two 5.25 drives just fine.
I will check after work.
Will also test Gotek on Amiga.

Ruben Corbo

keirf, 2021-11-05:
The issue can be voltage too high. There is an output chip 74HC04 which is not really correct to use with the MCU's output levels. The design just about gets away with it but not if Vcc gets too much above 5v (as this raises the 74HC04 input threshold even higher).

XGAMER777, 2021-11-05:
I got 5.30 and 12.54. It's also the Artery chip btw.

Ruben Corbo

keirf, 2021-11-05:
5.3 is probably okay. Bit high but others with problems have been ~6v. Try an Amiga I guess.

XGAMER777, 2021-11-06:
OK.

I've tried the Gotek in my Amiga 500 and it worked perfectly.

To summarize:
- The SX is not like other 1000s in that...
- Power does not come from the ribbon cable
- Power comes standard molex connector
- Voltage on the 5V is 5.3V
- My SX comes with two 360K 5.25 drives
- I'm plugging in the Gotek Tandy 1000 SX
- I'm using a straight through 34 Pin cable
- No twist just like the original Tandy cable
- The Gotek is the only drive being used...
- For the sake of simplicity during troubleshooting
- No cards or options in the SX
- Tried JC closed Select 0
- Tried JC closed Select 1
- Tried JC open Select 0
- Tried JC open Select 1
- Tried the FF.CFG template & appropriate options
- FF.CFG is in FF folder along with the other CFG
- Tried the IMG.CFG
- Tried multiple IMG boot images...
- From different sites
- The Gotek sees the file in the thumb drive...
- I can see it on the OLED along with the Track #
- The green access light flashes while trying to boot

It's driving me crazy as I feel it's something simple.
Could it be something to do with 360K format?

Ruben Corbo

keirf, 2021-11-06:
360K IMG should be interpreted as double-sided double-density 40-track-per-side. This is likely correct unless you have a much rarer single-sided 80-track system.

You don't need IMG.CFG at all unless you are explicitly configuring a special IMG geometry yourself. And then you would not be using the default one out of examples/ folder.

XGAMER777, 2021-11-06:
It's a standard 360K 5.25 drive.  Yeah... I was trying different things.  I tried with and without the FF.CFG and IMG.CFG.
It's obviously something specific with the IBM PC / XT setup and/or specific to this model Tandy 1000.
The drive works perfect on my Amiga 500.
I've seen YT videos of others getting it to work on previous models of the 1000, but nothing on the SX.
I may be the first to try.

keirf, 2021-11-06:
You might need to do something like put a logic analyser on it and see if the floppy signals are looking good. Eg rdata.

keirf, 2021-11-06:
I would also get rid of FF.CFG and reset cached config by holding both buttons a few seconds after power on but with no USB stick inserted.

XGAMER777, 2021-11-08:
Will do. Thanks.

Ruben Corbo

famiclone6502, 2022-04-24:
> I would also get rid of FF.CFG and reset cached config by holding both buttons a few seconds after power on but with no USB stick inserted.

I tried this but I am still having the same issues described by @XGAMER777 with my Tandy 1000 SX. I also tried the same troubleshooting steps recommended so far.

XGAMER777, 2022-04-24:
I got it working! I set the drive select to S0 on the Gotek and it is the only jumper installed. I installed a Card Edge to IDC Connector Adapter on the Gotek and connected the last connector on the Tandy floppy cable on the edge connector. But I had to flip the cable. I do not have a second drive connected as it's impossible with the short cable. I plan to make a longer cable with edge connectors and install a 5.25 drive as S1 on the middle connector. Also getting a longer molex to 3.5 floppy power cableDon't know yet if it will work. It reads 360K and 720K IMGs just find. It does NOT like 1.44 IMGs.

https://www.ebay.com/itm/163481202090

[image]

[image]

XGAMER777, 2022-04-24:
[image]

[image]

[image]

XGAMER777, 2022-04-24:
Installed MSDOS 5.0 from 720K images to a XT-IDE Lite Compact Flash card Hard Drive.

https://www.ebay.com/itm/224546590178
- Do not get the Lo-Tech XT-IDE... Does not work on the SX. 

The upgraded with MS-DOS 6.22 on 360K images. The images were downloaded for WinWorld.

https://winworldpc.com/product/ms-dos/1x

XGAMER777, 2022-04-24:
WARNING! If you get the compact flash card make sure to disable WRITING TO EEPROM as I have wipe two cards already.
I have purchased a EEPROM writer to rewrite the EEPROM. But this is a real pain.

XGAMER777, 2022-04-24:
I could not get it to work with a floppy cable with both 3.5 connector and the 5.25 edge connector and the twist at the end.
I may try that again at a later time. Maybe flipping the connector and setting both drive to S1 might work.

XGAMER777, 2022-04-24:
Actually... Just tried... Doesn't not work... Says cable is reversed. Tried several ways connecting the cable.

XGAMER777, 2022-04-24:
So... 
- Use a straight cable with edge connectors, NO TWISTS
- Use a Card Edge to IDC Connector Adapter on the GoTek
- Set the GoTek to S0, no other jumpers
- Use a Molex to 3.5 floppy power adapter
- Cable with twist and both IDC and Edge connector does not work
   (There may be some combinations were that cable will work, I could not get it to work)

famiclone6502, 2022-04-24:
> So...
>
> - Use a straight cable with edge connectors, NO TWISTS
>
> - Use a Card Edge to IDC Connector Adapter on the GoTek
>
> - Set the GoTek to S0, no other jumpers
>
> - Use a Molex to 3.5 floppy power [...]

Thanks @XGAMER777 for all the info and pictures! 

I've done these steps and removed my XTIDE just to be sure. I've tried several different disks including the one recommended. I see it trying to read but nothing happens. The green light might blink and/or I see 0_. This is the most life I've seen from it.

It could just be something wrong with the Tandy as I cannot get the disk drives to work either (I don't have any confirmed working drives to swap out and test with). I'm mainly trying to troubleshoot why the system always crashes in Tandy Graphics modes, and wanted to try loading something without the XTIDE OPTIMA card to see if it was causing the problem.

Thanks again, there was little info to go on for getting it to work on the SX. I think it's the Tandy motherboard at this point, not the gotek/flashfloppy.

XGAMER777, 2022-04-24:
Famiclone6502... I thought it was the Tandy mobo too at first. But I'll bet it's not. I have a math co and a NEC V20 in mine. Working just fine with the GoTek and XT-IDE lite with BIOS 4.1 (I think).

XGAMER777, 2022-04-24:
Are all the Dip Switches set to on the mobo?

famiclone6502, 2022-04-24:
>   Are all the Dip Switches set to on the mobo?

Near the processor? It's on off off off for 1234 respectively.

No math coprocessor but it has a 286 upgrade card. I may need to try to downgrade and test without it.

XGAMER777, 2022-04-24:
Jumpers and Dip Switches:

[image]

XGAMER777, 2022-04-24:
I would definitely remove the 286 card... And ship it to me. ;-) Just kidding.
I know also that if there is a an I/O address or memory address conflict the GoTek would work.
Although that the floppy controller conflicting with other stuff in the SX.

XGAMER777, 2022-04-24:
Dip Switch 2 and 3 are the ones to watch I think. I have all set to ON.

[image]

famiclone6502, 2022-04-25:
> Dip Switch 2 and 3 are the ones to watch I think. I have all set to ON.
> 
> 
> 
> [image]
> 
> 
> 
> 

Thank you so much for the suggestion, @XGAMER777 you've saved the day! I tried many combinations and all four dip switches need to be on for the gotek to work. 

I also was able to determine that the XTIDE really is causing the TGA crashes! The Tandy is fine, what a relief!

XGAMER777, 2022-04-25:
famiclone6502 That's awesome! Glad I could help.
Tandy Rulez!

closed 2022-04-25 by XGAMER777