keirf/flashfloppy #567: SFRKC30.AT2 DFU failure

quantum8, 2021-11-09:
I have a SFRKC30.AT2 board that I connected to with the artery tool and removed read protection on. I accidentally removed power before programming with flashfloppy and now the board doesn't go into DFU mode. Devicemanager lists the device as "Unknown USB Device (Device descriptor request failed).

The artery programmer won't recognise the board now. Is there another way I can force program the device with flashfloppy?

keirf, 2021-11-09:
You shouldn't be able to brick the device in this way but it sounds like what has happened. Check the cable, but otherwise: No, that's the only way to enter DFU mode.

closed 2021-11-09 by keirf

quantum8, 2021-11-09:
Well it turns out the USB port was dodgy. I tried again, but this time with the plug in only halfway and it worked!