keirf/flashfloppy #574: Korg I5M wont load

andr229k, 2021-11-21:
Hi,

### Enviroment:
•	Korg I5M with Gotek and firmware 3.29.
•	Korg I5M needs a 34 to 26 pin ribbon cable adapter, which I ordered from here:
https://www.ebay.de/itm/304011493903

### Symptoms:
I experience a handshake as the green led lights and I hear one click from the Piezo by disk actions, but it does not count-up on the tracks and is basically just says “Loading data” on the screen of Korg I5M.

### Diagnose:
The seller of the adapter told me, that his is how it works with the factory firmware:

_Hello, please download the program from the link: http://www.ketron.it/images/ketron/soft ... ketron.exe

1. Install the program as admin on the PC.
2. Execute the program (Virtual Floppy Format).
3.Select a flash disk (USB stick) min 8Gb.
4. Select number of partition: 1000
5. Select Floppy partition size 1.44 mb.
6. Start format.

After formatting start the program Virtual Floppy Explorer, select virtual floppy disks and copy midifiles.
The rest is self-explanatory.

Emulator SFR1M44-U100K has jumper settings. 1 jumper must be set to S1. And the others have to get out._

### Action:
Global -> from here I can choose my disks options [Load, Save and format etc..]

If jumper is on S0, then the device wont make any handshakes at all, therefore S1.

I have tested the following configurations, but with the same result as mention in Symptoms.

Test 1:
interface = Shugart
host = unspecified

Test 2:
interface = ibmpc
host = unspecified

Test 3: 
interface = ibmpc-hdout
host = unspecified

Any ideas or suggestions?

Note: I have attached two images, which I have made with GreaseWeazle from original Korg disks.
[Korg Tanzmusik 3 - I5S.hfe.txt](https://github.com/keirf/FlashFloppy/files/7576761/Korg.Tanzmusik.3.-.I5S.hfe.txt)

[i5si5mpl.img.txt](https://github.com/keirf/FlashFloppy/files/7576763/i5si5mpl.img.txt)

keirf, 2021-11-21:
Normal 720k PC disks I think. I would expect jumper S1 and JC (or set interface=ibmpc) to work.

The attached HFE looks corrupt on second side? No sectors found?

Are you using a known good image for boot?

andr229k, 2021-11-21:
I will give above a shoot with S1 and IBMPC [as JC-header is not featured on the newer Gotek designs]

If you open HxC Floppy Emulator Software -> Disk Browser -> 720kb DSDD FAT12 -> Create Disk -> Load Image and then not able to see the content as per screenshot below?
[image]

i5si5mpl is from here:
http://www.synthzone.com/midi/korg/i-series/i5s/i5si5mdl.htm

I did same as above and dumped the core files from the .zip into the browser window and exported as .IMG.
[image]

andr229k, 2021-11-21:
and an overview of those data on the disks from the manual:
[image]

closed 2021-11-28 by andr229k