keirf/flashfloppy #580: Help  Gotek Oled display is upside down.

stevehdj, 2021-11-24:
Help,  I have inserted a usb stick with std FF.cfg file and now display is upside down.  Before that all was normal using hxc software.  I have retried with std ff.cfg file but display remains upside down.  thanks

keirf, 2021-11-24:
You need the following display-type line in FF.CFG:
```
display-type=oled-128x64-rotate
```

stevehdj, 2021-11-24:
Keirf,
Thank you for your advice this has corrected my problem.  May I ask you a question.  I am using the drive on a Roland AT organ and have been using the hxc emulator software to load  files to USB stick. As you probably know this  saves files to the .HFE extension all of which is an added step in the process of getting files onto USB stick as opposed to copying the original files directly.  Also I have been having some issues writing to the USB from the organ.  With flash floppy firmware is there an way to load native Roland files directly on USB stick as it would be with a normal floppy. I have been looking at the info for FF on Github site but an not the most experienced when it comes to all this.  Thanks for your suggestions.  

Steve

keirf, 2021-11-24:
Supply an hfe image and native Roland file here and I'll have a look

stevehdj, 2021-11-24:
Emails keep getting rejected due to file size,  I will send separate emails

stevehdj, 2021-11-24:
HFE file keeps getting block for size,  the master image disks are  set to 39mb so cant get one to you .  So i had to make a small test hfe file.  thanks

keirf, 2021-11-25:
You have to upload and provide a link or attach them using the GitHub web UI. Email attachments get stripped out.

stevehdj, 2021-11-25:
Hi,
Below is link with a few of the native Roland files that are contained within the hfe file.  Hopefully you can open and thanks for continuing to help me with this. 

https://drive.google.com/drive/folders/1nmpyCFZytx1D4cSLAdHAL6W9_OalFjO1?usp=sharing

Miket277, 2022-01-09:
Steve, I am a fellow Roland AT organist and I bought a GoTeK floppy emulator, but I can't get it to work.  I tried jumper settings recommended by somebody, but it appears GoTek has been changing processors

stevehdj, 2022-01-10:
> Steve, I am a fellow Roland AT organist and I bought a GoTeK floppy emulator, but I can't get it to work. I tried jumper settings recommended by somebody, but it appears GoTek has upgraded its processor and jumper setting don't do the trick. Would you be willing to help me by email? Thanks. Mike. [pinecone277@gmail.com](mailto:pinecone277@gmail.com)

Hi Mike 

Which version Gotek do you have ,  I am using flash floppy and I seem to remember there was something in the setup that I had to change for USB stick to be recognized.   I will go through my notes later tonight and if I can find some info I will send later.   Also are you using the correct software to convert files,  the USB drive cannot read a Roland sty file directly,  has to be converted.  Attached is the program I use but again for Flash Floppy drives.
[image]

Later
Steve

stevehdj, 2022-01-10:
Hi Mike 

Which version Gotek do you have ,  I am using flash floppy and I seem to remember there was something in the setup that I had to change for USB stick to be recognized.   I will go through my notes later tonight and hopefully give you a better answer.  Also are you using the correct software to convert files,  the USB drive cannot read a Roland sty file directly,  has to be converted.  I believe the program can be downloaded from the Github site

Later
Steve

closed 2022-01-10 by stevehdj

Miket277, 2022-01-10:
Steve --

Thanks so much for helping me.

I got a "new" Gotek model SFR1M44-U100 which has the "AT" Artery chip.  I have since learned that the models up until around March, 2021 had an "ST" chip. 

Here's what I've done so far:

1) I disassembled my organ, a big job, with screws and parts all over the living room.

2)    I installed the emulator drive, got it to fit nicely, and was so proud of myself that I put everything back together prematurely.

3)    When it came on, I had  "0.0.0" on the little screen, just like the directions said.  But when I inserted a USB stick, the periods did not disappear so I knew something was wrong.

4)    It would not read a USB stick, and would not format a USB stick through the Roland method (Utilities, Disk Utilities, Format . . . etc) and it would not format the USB Stick through the Gotek method (turn power off, insert stick, hold down two buttons).

5) I got back on Google and found advice about jumper settings, especially having the MO (motor on) jumper set.

6) I took the organ apart again and tinkered with the jumpers in all possible combinations to no avail.

7) I found "Flash Floppy", downloaded it and successfully flashed the drive.   Now when you turn it on, you get "F-F" on the little display.

8)    I found advice by GoTek that the only jumper setting on the AT chip model necessary is one jumper in either S1, S0 or MO and none others.  I tried all three one at a time, and still no dice.

9)  I made sure the ribbon cable is right-side up (it reads "RIB" when its upside down).

I'm totally stuck at this point.

If any of this jogs your memory, I'd appreciate your advice.

Thanks so much for your help.

-- Mike

keirf, 2022-01-10:
What images are you using on your usb stick? Does display change when you insert the usb stick?

Miket277, 2022-01-10:
I tried two images from Rufus for a about 6 720 disks and four 1.44 disks, but they won't load.  They remain at "F-F" when inserted.  When I try to format them in the organ, I get an error code, I think E32 (I'm at my office).    Thanks.

keirf, 2022-01-10:
FlashFloppy works with IMG files on a FAT filesystem on the usb stick. This is why display doesn't change when you insert the stick: FlashFloppy doesn't find a FAT filesystem. You need to get FlashFloppy to recognise an image file.

Miket277, 2022-01-11:
Thanks for your advice.  I am still stuck.  I used Rufus to flash the USB stick with a FAT32 free DOS bootable image, no luck.  I tried Etcher to load IMG files from your alt folder (a 720 and a 1.44) still no luck.  I'm getting "E34" when the organ tries to read the USB, and "E01" when the organ tries to format the USB.  Where do I find the right IMG file to flash, and what program should I use to flash the USB stick?  Thanks for your help.  I'm (obviously) not a computer wizard, I'm a Sunday church organist trying to get my Roland Atelier Organ up to date.  Thanks. Mike Tremblay

keirf, 2022-01-11:
Here is your confusion: You do not flash your image direct to the USB stick in this way! Just format your stick empty FAT32. It's now just a normal, empty, USB stick (like it was fresh from the factory). Your images are IMG files copied **as files** onto the USB stick. FlashFloppy will read the FAT32 filesystem, find the IMG files, give them arbitrary numbers (for now, I can explain how to do better later) and let you select between them.

Does that make sense? Stop Rufus'ing your stick!

Miket277, 2022-01-11:
Aha!  I did as you said - formatted the USB stick and copied an image file into the root directory.  I used DSKA0001_Roland.QD.  Once loaded into the organ, I noticed the "F_F" display was gone and replaced with "9-d" when the organ boots up.  But the display changes to 000 right away when the USB stick is inserted.  However, that's as far as it goes.  The Organ won't read the disk or write to it.  And I noticed a file called IMAGE_A.CFG was created in the root directory on the USB flash drive.  I seem to be making progress, but it's still not working.  Thanks for your help.  Mike.

keirf, 2022-01-11:
If you are using an image with suffix QD then the Gotek is replacing a QuickDisk drive, yes? And you are using the FlashFloppy QuickDisk firmware?

Miket277, 2022-01-11:
No.  I don't think so.  I do not know what a QuickDisk drive is.  I tried several image files, and the .QD file was the only one I could find that was read by the Gotek.   I tried it because it had Roland in the name.  Should I be using something else?

stevehdj, 2022-01-11:
Mike,  if you want to PM me on Facebook I will send you some files to try that I know are working fine on my 80SL

keirf, 2022-01-11:
Perhaps it's not .qd and has a hidden extension/suffix. If the Gotek says F-F at power on then it's not QD firmware and shouldn't accept QD images anyway. So perhaps that's all a red herring. In any case getting a working image and the Gotek correctly jumpered is your next hurdle.

Miket277, 2022-01-11:
After I first flashed Flash Floppy, it did say "F-F" when it booted up.  Now, somehow, it says "9-d" whatever that means.  Could I have installed an update accidentally?  I though .UPD files might be images and might have loaded one by accident.  Should I re-flash Flash Floppy?

Miket277, 2022-01-13:
Can somebody point me to an IMG file that works on a Roland Atelier organ?
I have Flash Floppy installed on the GoTek Emulator with the AT chip, but it reads E34 when I insert a USB stick.  Thanks.  Mike.

keirf, 2022-01-13:
Here is a PC-formatted 1M44 image. Just needs to be unzipped to USB stick.
[1m44_formatted.zip](https://github.com/keirf/flashfloppy/files/7861317/1m44_formatted.zip)

keirf, 2022-01-13:
And here is 720k PC formatted.
[720k_formatted.zip](https://github.com/keirf/flashfloppy/files/7861794/720k_formatted.zip)

stevehdj, 2022-01-13:
Mike,  I sent you a couple of files yesterday via your personal email,  if you didnt get them let me know,