keirf/flashfloppy #583: Gotek Artery design / QD and JC

andr229k, 2021-12-09:
There is mentioned in the following documentation for Quickdisk that Roland devices requires a jumper on JC:
https://github.com/keirf/FlashFloppy/wiki/Quick-Disk

What do you do, when you don’t have a JC on the new Gotek Artery design?

keirf, 2021-12-09:
No workaround on current QFN32 models. If really needed I could probably implement alternative location at PA13 on prog header

Nicky831, 2021-12-22:
I also miss the "JC" jumper on the new Gotek SFRKC30.AT2.
Now it doesn't work in the Roland 220.
What exactly does jumper "JC" do? Can it be adjusted with an entry in the "ff.cfg"?

keirf, 2021-12-22:
It slightly tweaks the behaviour with regards to the MOTOR signal. There is no way to adjust it via FF.CFG, at least currently.

keirf, 2022-02-19:
@Nicky831 are you interested in testing alternative locations for jumper JC on AT2 Gotek boards?

Else I will close this ticket due to lack of ongoing interest.

Nicky831, 2022-02-20:
Yes I am interested. At the moment I don't have a Roland machine to try it out. But I can ask the buyer of the machine if he wants to test it. I'll get back to you soon.

Nicky831, 2022-02-21:
I have received feedback. I can do the test and share the data/results.

keirf, 2022-02-21:
@Nicky831 well here is a start. Current firmwares will presumably not work with AT2 boards since the Roland needs JC jumper. So attached here are two firmwares with JC behaviour forcibly enabled. So it would be interesting to confirm that vanilla firmware does not work, but these modded firmwares do. If that checks out, then I can do a firmware that implements JC on the programnming header.
[JCForce.zip](https://github.com/keirf/flashfloppy/files/8111274/JCForce.zip)

shea1337, 2022-04-22:
Hi, I'm also curious about this, so I am working on it.  I have for example a Roland MKS-100, in my possession,  which I suppose needs the JC jumper signal.  Unfortunately, the error it gives for an invalid signal from the disk drive is the same as the error it displays when the QD is blank. I have some sounds as Sysex files, so I'll get a bank sent over MIDI and attempt a save. 

I have 4.99a from the above zip file flashed at the moment.

shea1337, 2022-04-23:
Is the track display supposed to be active with the QD firmware? I am getting T:00.0 on all my write / read attempts, although it's mostly working.

AT2 Gotek, jumper only on MO, saving to the default "Blank.qd" as provided alongside the Flashfloppy binaries:

 Saving: works as expected however displays an untrue verification error. (Display reads "Check QD", informs you if the disk contains a different bank that will be over-written, press the ENTER button to verify, it saves the bank, then "Verify Error" - but the file does save.)
 
 Loading: works as expected without errors.
 
 
@keirf  What could I check out to see if we could prevent that "Verify Error" from happening? I do have access to a logic probe.

shea1337, 2022-04-23:
No idea if this will assist on debugging, so attached is a Roland S-10 format QD image I verified loaded without issues after saving, despite the "Verify Error" warning.
[Roland S10 Elec Organ 1.zip](https://github.com/keirf/flashfloppy/files/8546201/Roland.S10.Elec.Organ.1.zip)

keirf, 2022-04-23:
The saving probably takes some time and the host is waiting to be able to read back for verify and times out. Perhaps try another usb stick. Does the normal firmware also work or is the special firmware here actually needed?

shea1337, 2022-04-23:
> The saving probably takes some time and the host is waiting to be able to read back for verify and times out. Perhaps try another usb stick. Does the normal firmware also work or is the special firmware here actually needed?

I am running the normal 3.30 Stable firmware on an AT2 Gotek this Roland and I can still save and load fine.

One issue I have found today is that I am not able to load multi-image banks on either the normal or special firmware. Some of these banks are large enough that they must be saved to both sides of the Quick-Disk. When the Roland prompts me to flip the disk, I switch to the next image and wait for it to detect and start saving. However when loading, it will give an error "I/O Error 2", which the manual doesn't describe in any detail. Note that  I am not having any issues with banks which fit in a single QD image (banks that do not prompt for the disk to be flipped when saving or loading).

My old 2.5" drive is still here and working, and I could load banks where you need to load both sides of the disk without issue, so I think I can rule out a problem with my sampler.

I am still getting "Verify Error" after the save is done with either a Sandisk Cruzer Blade USB 2.0 or a Kingston USB 3.0 drive. I also tried setting both track-change and write-drain to "realtime", with no change, so I set them back to "instant" as they were.

I will attach a set of two disk images which showed the typical "Verify Error" after saving but gives the I/O error when attempting to load. It always shows that I/O error shortly into reading from the 2nd image. [Wurlitzer - not sure if good.zip](https://github.com/keirf/flashfloppy/files/8546433/Wurlitzer.-.not.sure.if.good.zip)

keirf, 2022-04-23:
The multi image issue is weird. Maybe worth running quickdisk logfile firmware and gather FFLOG.TXT file.

shea1337, 2022-04-23:
> The multi image issue is weird. Maybe worth running quickdisk logfile firmware and gather FFLOG.TXT file.

I put on the logfile firmware, loaded the bank via MIDI, and saved it to two empty disk images, and then cleared out the internal memory and attempted to load it again. I got an Illegal QD error immediately upon trying to load unfortunately which seems to be isolated to the logfile firmware.   I did re-flash and double check I didn't do anything wrong or differently.  I then loaded a known good single-image bank and that worked fine. 

I will have to try out more experiments with stock & modified firmwares on this multi-image problem. 

I don't know what the issue would be but going back to non-logfile QD 3.30 firmware, I can at least get the first image loaded and the second image partway loaded before giving the "I/O Error 2" (QD drive related error).  For whatever reason with the logfile 3.30 QD firmware it immediately said "Illegal QD" trying to load these same images.  The single image bank that did work loaded properly.

[Two freshly made images containing the same data as the above "Wurlitzer" and the log file](https://github.com/keirf/flashfloppy/files/8546473/two.images.and.log.zip)

Let me know if I can do anything to help.

shea1337, 2022-04-26:
Little update, I have a couple multi-image banks working, verified still working after I reboot the unit.
I've made a few dozen disk images, some single image and some multi image, saved and reloaded them all, verified no corruption and reliable performance.

**- I need to save with 3.99a specifically (JCForce firmware)  for 100% reliability** 
**- I need to wait an extra ~10 seconds before I switch the disk image, instead of switching right when it prompts me to flip the QD or very shortly thereafter (waiting for writes to complete I guess?) or I'll get the dreaded I/O Error 2**

I would consider the JCForce firmware to be necessary for this Roland and most likely others since a few of these QD-based Rolands share certain hardware. 3.99a specifically is working best. I still have not found the root of that peculiar "Verify Error" which could still come down to an issue with my USB stick, but not 100% sure.  I am not getting that error reliably or even often on the modified firmware.

I might see if i can make some tweaks to the QD firmware code that gives some sort of indication of what % of the disk is written to, and maybe to delay reporting back to the host that the write is complete just a bit, not sure if that is anything of interest.

shea1337, 2022-04-26:
The tl;dr of all that is that it's mostly working as expected with the alternate firmware, and if I could get a logfile build of JCForce 3.99a to mess with then I could most likely figure out why I still get a corrupted save once in a while.

keirf, 2022-04-26:
Ok, 10 seconds is a long time that can't really be writeback unless the USB stick is very very bad. It's not really plausible. Perhaps it's to do with the QuickDisk emulation playing out to the end of the "track" and stopping. Which is what a QD drive does when the motor signal is stopped. Maybe it's a bug in emulation of QD eject/insert. There's some options here.

keirf, 2022-04-26:
There isn't a spare MCU pin to move the JC jumper to for SFRKC30.AT2. Instead let's make use of `interface=ibmpc` in FF.CFG to mean JC strapped on SFRKC30.AT2 running QD firmware. A bit of a hack but good enough eh? Here is a test firmware which implements this, for you to try (you need to be logged into github to see the download artifact at bottom of page):
https://github.com/keirf/flashfloppy/actions/runs/2226213895

keirf, 2022-04-26:
PPS. Would this workaround be more generally useful do you think? Should I respect `interface=ibmpc` on all Goteks running QD firmware?

shea1337, 2022-04-26:
> PPS. Would this workaround be more generally useful do you think? Should I respect `interface=ibmpc` on all Goteks running QD firmware?

I can't seem to imagine a reason not to do this. As far as I know, this behavior of needing the JC signal set only affects Rolands. Funny enough you mention this as I actually tried setting `interface=ibmpc` to trick the stable firmware into "seeing" the JC signal before opening this ticket, so others would probably also come to the conclusion that they may be interchangeable settings if I did.  I'm a fan of settings that can be done via the text file anyway ;-)

I'll get back to you after I have some time to mess with that build you just made (which I do have downloaded now). Hopefully I can get my hands on some more QD gear soon to do tests on.

keirf, 2022-04-26:
This build makes the `interface=ibmpc` alternative available on all Gotek models:
https://github.com/keirf/flashfloppy/actions/runs/2226358356

keirf, 2022-05-04:
I think that wraps up this ticket. Next release will include the JC strap alternative `interface=ibmpc`. If there are further issues to discuss let's use a new ticket.

closed 2022-05-04 by keirf