keirf/flashfloppy #584: suggested software for flashing Artery under Linux?

DrYak, 2021-12-11:
Hi!

The [linked post on MSX forum](https://www.msx.org/forum/msx-talk/hardware/goteks-with-new-guts) only mentions [Windows software for flashing](https://www.arterytek.com/download/Artery_ISP_Programmer_V1.5.46.zip)  Artey chips.

On Linux I have tried using dfu-util v0.11, but it fails at the unprotection step with:
```console
dfu-util: Wrong state after command "READ_UNPROTECT" download
```
Is there a suggestion for successfully flashing from Linux? Or trying to dust out some Windows VM and forwarding it the USB port is the only solution?
Thanks!

keirf, 2021-12-11:
I haven't tried it I'm afraid. My guess is that dfu-util would need patching.

ejona86, 2021-12-13:
I successfully flashed an Artery from Linux. But it turned out easy enough I didn't write it down... I have dfu-util 0.11. I think it involved "ignore the error message" and maybe "power cycle the Gotek." I might have also used lsusb to determine the device for `--device`.

BakedSnake, 2021-12-16:
I get the same "READ_UNPROTECT" on linux. anyone has more info on how to flash on linux?

keirf, 2021-12-16:
Beased on ejona86's response I would try ignoring the error, reconnect the Gotek, and try the line
`sudo dfu-util -a 0 -D FF_Gotek-v3.*.dfu`

prosper00, 2021-12-16:
I just used the UART / serial bootloader. Worked great. Self-updates from the USB stick work well too to keep things updated.

BakedSnake, 2021-12-16:
So, I ignored the error and continued with the next command. After that I had to update my gotek with the upd file and everything works.

I was able to boot my mpc2000.

So I guess just dfu-util is enough.

Thanks

conloos, 2021-12-26:
Hello @all,

I had some problems to flash the firmware with a USB-TTL. 
Following my solution based on a thread at the [Amiga forum](http://eab.abime.net/showthread.php?p=1175795).

My environment is: Ubuntu 21.04 with stm32flash 0.5 and a USB TTL from D-SUN.

First the read protection has to be switched off:
$ sudo stm32flash -k / dev / ttyUSB0
Then clear the memory:
$ sudo stm32flash -o / dev / ttyUSB0
Finally, the firmware can be flashed without any problems.
$ sudo stm32flash -vw FF_Gotek-v3.29.hex / dev / ttyUSB0

Switching off the flash write protection was acknowledged with an error:
$ sudo stm32flash -u / dev / ttyUSB0
stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1
Version: 0x32
Option 1: 0x46
Option 2: 0x01
Device ID: 0x0410 (STM32F10xxx Medium-density)
- RAM: Up to 20KiB (512b reserved by bootloader)
- Flash: Up to 128KiB (size first sector: 4x1024)
- RAM option: 16b
- System RAM: 2KiB
Write-unprotecting flash
Got NACK from device on command 0x73
Failed to write-unprotect flash

hugoduraes, 2021-12-29:
I wonder if using dfu-utilon a mac would work too! 🤔

coredump4, 2022-01-30:
I would love it if it worked on Macs.  Under macOS Monterey, with dfu-util v0.11 and libusb v1.0.24 built from Mac Ports, I got this when trying to erase an STM32F105:
`dfu-util: dfuse_download: libusb_control_transfer returned -7 (LIBUSB_ERROR_TIMEOUT)`

hugoduraes, 2022-01-30:
It actually worked for me with the same errors on the 1st command, as described above.

Olas-Audio, 2022-10-01:
Hi everyone. A few years ago, I got an Agilent 54622A oscilloscope for about 100 USD, a real bargain and the truth is that it is an excellent measurement equipment, despite being just 100 MHz. I just want to share the methodology that I followed to be able to use a Gotek floppy emulator. Without having read this issue, I decided to buy the cheapest emulator of this brand (34 USD shipping included), it’s an SFR1M44-DU26 SFRC2D.B version. So, there is no need to buy an expensive type to perform this operation.

Step 1. Download FlashFloppy version 3.29 (I think 3.3 could also be used).
https://github.com/keirf/FlashFloppy/wiki/Downloads

Step 2. Download the file Artery_ISP_Programmer_V1.5.46 or later and install the driver “Artery_DFU_DriverInstall.exe”.
https://www.arterytek.com/en/product/AT32F415.jsp#Resource
https://www.arterytek.com/download/TOOL/Artery_ISP_Programmer_V2.0.06.zip

[Artery_ISP_Programmer_V1.5.46.zip](https://github.com/keirf/flashfloppy/files/9691279/Artery_ISP_Programmer_V1.5.46.zip)

Step 3. Prepare the PCB for flashing according to the follow images:

[image]
This version uses the Artery microcontroller.

[image]
SFRC2D.B version.

[image]
I used a USB type A cable on one end and on the other side I exposed the wires so I could solder them to the PCB. I performed this step since I didn't have a USB A-A type cable.

[image]
I soldered the wires in the shown sequence to the base of the USB connector on the PCB, the layout is as follows:
|GND-black|D+ Green|D- White|+V-Red|

[image]
I soldered a 5-pin header to put a jumper between Boot-3V pins. I only left the S1 jumper in place.

[image]
Install the driver.

[image]
Then run the app “ArteryISSProgrammer.exe”.

[image]
Select USB DFU as the port type. Then we proceed to connect USB cable.

[image]
Select language and click next

[image]
Click next

[image]
Click next

[image]
You must select Enable/disable protection, click yes from the dialog box and then click next

[image]
Wait until it’s done the click back

[image]
Click Download to device and then click add

[image]
Look for the Hex file of the version you want to install, in my case 3.29.

[image]
Select Global erase and verify after download then click next

[image]

[image]
Finally click close. You can unplug the USB cable from the computer.

Now you can desolder the resistors RA, RB and RC. The memory IC does not need to be preserved. I removed it so I could use it in some other project later on.

[image]
[image]
Don't forget to remove Jumper Boot-3.3V.

Finally, it is necessary to prepare you USB device. Give it a format FAT32 and then you need to copy the "1m44_fat12.img" file into the main directory of the USB that you just formatted.
https://github.com/andrewpono/TDS3000-Series-Gotek-FlashFloppy/blob/main/1m44_fat12.img

That’s it, your scope should be ready to save Images into your USB stick. The file storing time it’s slow but it’s definitely more convenient than to use a Floppy Disk.

Cheers and enjoy your measurement!

closed 2022-12-19 by keirf