keirf/flashfloppy #585: Enhanced Gotek and Artery

jasonsbeer, 2021-12-13:
Greetings - 
I am working on a Gotek PCB project based on the Artery AT32F415RBT, which is found on the factory Gotek SFRKC30.AT4. I want to include your "Enhanced Gotek" features in my design. However, this board is wired differently than the original Gotek with the STM32F105 MCUs (reference Gotek schematics from H.M.).

Signal|STM32F105 Pin|AT32F415 Pin
-|-|-
JA|PC6|PA5
S3|PC7|PA4
S4|PC8|PA3

Your planned enhancements call out PA3, PA4, and PA5 as reserved for those enahcements. As you can see, none of these pins are available if I follow the factory signal routing. **My initial intent is to maintain the factory pin outs in my project**.  I assume you are already aware of these differences because you would need to account for the S3 and S4 differences in your current firmware.

So, some questions...
1) Is there a way to maintain enhanced gotek compatibility with the AT32F415RBT, considering PA3-5 are not available?
2) Can I connect the signals in the same way as is is done on the STM32F105 boards?
3) Do you plan to standardize a specific MCU for the "enhanced" features?

BTW - great work!

keirf, 2021-12-13:
I suggest you stick to the original schematic implemented onthe old STM32 boards. These are actually fully compatible with AT32F415RBT (in recent FlashFloppy firmwares). You can then also implement the enhancements.  The enhancements are not supported on any of the recent AT32-specific schematics.

jasonsbeer, 2021-12-13:
Much appreciated.

closed 2022-01-10 by keirf