keirf/flashfloppy #588: Artery Gotek Fails to Read/Format Double-Sided Disk Image (TRS-80 LDOS)

artagesw, 2021-12-23:
New Gotek/FlashFloppy user here, trying to get it configured and working with a TRS-80 Model 4 using LDOS.

FlashFloppy v3.29 successfully flashed, and I can format/read/write 180K (SSDD40) and 360K (SSDD80) disks just fine (.hfe images). But I have not been able to get a 720K (DSDD80) disk to format. Have tried both .hfe as well as custom raw .img images.

I even found a known good LDOS 720K .hfe image (from a TRS-80 archive site), and although FlashFloppy recognizes the disk, it returns read errors when attempting to access its directory (DIR command). Attempting to reformat that .hfe image fails in the same manner as my own earlier attempts.

The formatting appears to proceed normally through all cylinders. But subsequent verification fails on every track, aborting the format operation.

I've never owned the previous STM-based incarnation of Gotek drive, but it seems other TRS-80 users have previously had success with DSDD80 720k images with their Goteks, so I am wondering if this may be an Artery-specific issue.

keirf, 2021-12-24:
Worth trying v4.3a firmware and see if that works.

artagesw, 2022-06-22:
Sorry, I should have closed this issue. It turned out to be a bad cable, not a software problem. Thanks for checking back (and for the amazing software).