keirf/flashfloppy #590: Rotary support on SFRKC30.AT2 - alternative pin location

StereoRocker, 2022-01-02:
Hello,

I have 2x SFRKC30.AT2 model Goteks. They have the pin block labelled JK/J8 where I can connect rotary encoder, but on one of my models the DT pin doesn't work - it was filled in with solder and I broke the pad while desoldering.

However, I also notice the full programming header is on the board, so PA13 and PA14 are exposed. I have followed the traces on the board and confirmed against 32-pin AT32F415 pin diagram to be sure of this. However, on this model, flashfloppy doesn't seem to respond to a rotary encoder on these pins. I see in `board.c` it only reads PA13 and PA14 if it detects a 48 pin MCU, but will override it anyway if KC30 header is detected. Could a configuration option be added to instead look at PA13 and PA14 as an alternative on boards, maybe even specific to KC30, where those pins are available please?

I am of course happy to help test any changes.

keirf, 2022-01-04:
No, just repair your pad! XD

keirf, 2022-01-04:
A bit more advice: Solder header pin block to JK/J8 except the broken pad. Scrape some solder mask off the trace leading to broken pad. Flux and solder fine wire to that trace (eg from IDE PATA 80 ribbon). Wrap the wire a couple of times round the pin that pokes through the broken pad. Now apply solder to that pin and the wrapped wire.

closed 2022-01-04 by keirf

StereoRocker, 2022-01-04:
Hah, yes, repair is on the agenda. I need to get different pin headers - the ones I've soldered aren't right angled so the wires don't fit lol.

I actually did amend the source and rebuild locally so I can use the drives until I have the necessary bits, and did so by adding a config option. If I were to submit a PR would you consider including it?

keirf, 2022-01-04:
No, it's confusing enough as it is. If it has the KC30 header, that header must be used.