keirf/flashfloppy #591: Dynacord ADD two - Loading Akai S-900 disks.

andr229k, 2022-01-02:
In 1990 Dynacord released the ADS ADD-two sampler and the strong sales point here was the ability to read Akai S-900 disks as these was a studio standard equipment sampler at this time. 

Im able to load Dynacord Factory disks without problems, but I have problems with Akai S-900 disks.

**Environment:**
ADS ADD-two sampler installed with a Floppy Emulator with Artery-MCU and Firmware 3.29.
interface = ibmpc-hdout
Jumper -> S0
host = unspecified

**Symptoms:**
when I try to load Akai samples from a .IMG disks, then it counts up to 204 and give me the error “Error! Diskette is unreadable”, but I can see that it has loaded something into the memory as the sound can be found but not being proper playback.

I made a video of it here:
https://drive.google.com/drive/folders/15Lmhr6l_tv1LHMqwDP2MX8Bb7jbsmMrj?usp=sharing

**Diagnose:**
•	Since we are able to load Dynacord Factory disks, then we must have done something correct in regard to the configuration file.
•	Akai S900 only used DD disks.
•	Akai S-900 disks cannot be loaded directly, therefore I had to go into a sub-menu to find the “importer” for .SMP samples, which is used for loading Akai S900 disks.
•	A user one a German forum mentioned, that he was able to loading disks, but he was using firmware version 2.13 [I cannot downgrade due to Artery-MCU]
•	A DD disks does not have 204 tracks [Normally max 84 tracks]

**Actions:**
N/A

Any idea or suggestions?

[image]

andr229k, 2022-01-03:
Issue resolved with HxC Floppy Emulator Software conversion.

closed 2022-01-03 by andr229k

chickeneps, 2022-12-17:
I'm having issues reading images (made with OmniFlop) of the regular Dynacord library.  IMG files, 1600K. Same Environment as you mention above. I get the DISKETTE IS UNREADABLE with that even.

It's funny that would even work, since OmniFlop sees Dynacord as 80x2x20 512. This means there's 10 sectors per side, not the usual 9 sectors, so I don't know how your environment would work.

Any suggestions?

keirf, 2022-12-17:
80x2x20 suggests 20 512 byte sectors per track doesn't it. This could be supported out of the box. Have you tried interface=ibmpc-hdout?

To get Akai DD disks working would require IMG.CFG as host=akai in FF.CFG would change interpretation of 1.6MB image files to have 80x2x10 1kB layout. Hence IMG.CFG would be needed to define new geometry for 800kB images only.

chickeneps, 2022-12-17:
So I'm going to try a IMG.CFG with this, does this sound right? Does the tag seem right, in it would apply to all .IMG files that are 1638400 bytes large on the USB drive?

[::1638400]
cyls = 80
heads = 2
secs = 20
bps = 512

This didn't work though. Attached is a HFE file that does work, can you suggest a proper IMG.CFG?
[112.006 CANNON CARS.zip](https://github.com/keirf/flashfloppy/files/10252086/112.006.CANNON.CARS.zip)

keirf, 2022-12-17:
Try the attached IMG.CFG. The sectors are interleaved 2:1, and the sector numbering continues from head 0 onto head 1 (head 0 has sectors 1..20, while head 1 has sectors 21..40).

[IMG.CFG.zip](https://github.com/keirf/flashfloppy/files/10252184/IMG.CFG.zip)

chickeneps, 2022-12-17:
It works! Thank you so much, as usual. You're a hero. (Actually I'm a hero, ask any Swedish person =) REALLY) - but thanks. Add it! The Dynacord ADS is also called (although not printed) the ADD-TWO, but just add it once. =)
But seriously, this relates to the Dynacord ADS and ADS-K, not necessarily the ADD-ONE, but it probably applies.

chickeneps, 2022-12-17:
Just a unrelated/but/related followup, why can't a custom IMG.CFG be written for the Emulator II, Ensoniq Mirage, and SCI Prophet 2000/20002 disk layouts? I'm big on getting away completely from HFE files.

keirf, 2022-12-17:
Emu2 does not use IBM track format, so it is out of scope for the IMG handler. It will probably never be supported natively.

Mirage has mixed sector sizes. Not yet supported. Possibly Prophet too.

If you supply HFE examples for the above I will see how easy it is to extend the IMG.CFG parser to support them (the IMG handler is capable internally, just needs to be describable in IMG.CFG).

chickeneps, 2022-12-17:
Here are a Mirage HFE and a Prophet HFE. I think they are related, maybe the same, perhaps slightly different.
[MIRAGE.HFE.zip](https://github.com/keirf/flashfloppy/files/10252385/MIRAGE.HFE.zip)
[PROPHET.HFE.zip](https://github.com/keirf/flashfloppy/files/10252386/PROPHET.HFE.zip)

keirf, 2022-12-17:
Interesting. Is Mirage really single sided? It has 5x1kB + 1x512B sectors per track. That makes 440kB (SS) or 880kB (DS) per disk.

Prophet is double sided with 5x1kB + 1x256B sectors per track. That makes 840kB per disk.

I could easily support these I think.

chickeneps, 2022-12-17:
Yep, Mirage used SS/DD disks. Once you couldn't get those, it was DS/DD using only one side. And yes, that's the sizes of the disks, I support writing both these images (and it was up to the HxC utility to get them into HFE).

chickeneps, 2022-12-17:
The 5x1kB + 1x256B sectors per track is where the Mirage and Prophet are alike. Mirage does 5 sectors of 1024 bytes and one sector of 512 bytes.

keirf, 2022-12-17:
So they're not exactly alike because one has 256b odd and the other 512b odd?

chickeneps, 2022-12-17:
Oh =) Time for new glasses

keirf, 2022-12-18:
Please try the following firmware and the attached IMG.CFG for both Mirage and Prophet. You need to be logged in to GitHub for the firmware download link to work. Ignore the debug zip file contained within and just use the normal firmware.

https://github.com/keirf/flashfloppy/suites/9932531120/artifacts/480727222

[IMG.CFG.zip](https://github.com/keirf/flashfloppy/files/10253866/IMG.CFG.zip)

chickeneps, 2022-12-19:
Thanks; I haven't tried this yet, but I want to say that there isn't a DS/DD version of Mirage storage - it's just that the Mirage has a single-sided drive and doesn't ever read the "other side" if it's a DS/DD disk. Same coercitvity.

chickeneps, 2022-12-20:
Happy to report they load in great - thank you!