keirf/flashfloppy #596: Latest GOTEK Firmware 4.1a - 4.4a. ERROR ... LBA Change test failed! Write issue?

Tallguy297, 2022-01-14:
I flashed my Gotek with the latest 4.4a firmware.
When I reset my Amiga and used the Software selector I received this error...
`ERROR! LBA Change test failed! Write issue?`
I have reflashed the Gotek with previous firmwares from 4.1a - 4.3a. Same error each time.

The only time I did not get this error is with firmware 3.29.

lordoftears, 2022-01-23:
I am experiencing the same identical issue

andiweli, 2022-01-23:
Same experience with latest 4.4a on a GoEX drive. 
Booting from preconfigured ADF disks on my A1200 works, only AUTOBOOT.HFE/FileSelector refuses and shows this bug.

lordoftears, 2022-01-23:
I did some tests. I can launch the adf files I already had in autoboot, but I cannot boot autoboot itself (same error as in topic title). Last firmware working is 3.29

Emanuele

keirf, 2022-01-24:
There is no need for further user testing. It is definitely borked! :)

ejona86, 2022-01-30:
@keirf, did you take a look with a logic analyzer? A logic recording would be useful if it is easy.

It'll probably be a week more before I'm able to try to reproduce by adding code to ff-testbed. I've read through my async porting commit and don't see any obvious bugs, especially for SET_LBA which I didn't change. That has me figuring this is likely a timing/buffering issue, but I'd be happy if it were a simple bug instead.

keirf, 2022-01-30:
@ejona86 I have not found time to investigate as yet. This may require repro and serial debugging with an actual Amiga, as of course testbed does already make use of DA mode with no problems.

Some people have seen this error on Amiga AUTOBOOT fairly reliably from cold boot since forever. I never repro'ed that. With luck it could be the same bug somehow made 100% probable here (wishful thinking).

Liriel-666, 2022-05-30:
Same error on atari st 4.1-4.4a

Last Version that work on my gotek with  Artery  chip is 3.30

gearmesh, 2022-07-02:
I have it with 4.4a on an Amiga 1000

tacksetB, 2022-07-29:
Same bug on Amiga 500 with an AT32F415 Gotek and FF 4.5a pre-release

Had to downgrade the firmware to FF 3.34 to make my Gotek work again.

closed 2022-07-29 by keirf

keirf, 2022-07-29:
Okay I debugged this and it was a pretty simple issue to do with the write-protect line status. It's fixed now and I've released v4.6a

ejona86, 2022-07-29:
The bug was introduced in 1937c174d798f4ae42aa55f9abe55492fdb3e4c1

madk71, 2026-11-03:
Hi, I have the same issue with my Atari STF. I tried FF from 3,30 to 4.6a but always the same. A warm reset or two fixes it for a while but it feels Slow. Used 3 different USB drives (2x Kingston and 1 SanDisk); the biggest is 16GB, and the smallest is 8GB

commit 85611973 (2022-07-29): direct access: Clear the write-protect line when entering direct-access mode

Otherwise host command writes are ignored.

Fixes #596