keirf/flashfloppy #600: Casio FZ10M – Disk Error

andr229k, 2022-01-17:
**Environment:**
Gotek installed with Flashfloppy 3.29 and jumper on S0.

interface = Shugart
host = casio

**Diagnose:**
Above setup was tested with another floppy emulator, which worked without issues.

**Symptoms:**
Casio FZ10M is able to see the disk, but give the error “Disk Error”, when trying to execute load. [See screenshot]

**Actions:**
•	Tested different usb-sticks [problem remained]
•	Tried to install this Floppy emulator drive in another Casio sampler [problem remained]

Any ideas or suggestions to investigate, if the Floppy Emulator is “defect”? 

**Screenshot:**
[image]

**Demo image:**
[Cello_Violin.hfe.txt](https://github.com/keirf/flashfloppy/files/7884658/Cello_Violin.hfe.txt)

Dietrich-L, 2022-01-17:
I suggest to remove the .txt extension

andr229k, 2022-01-17:
> I suggest to remove the .txt extension

No, then you cannot upload it to GitHub.

keirf, 2022-01-25:
What other floppy emulator did you test? Another Gotek? With FlashFloppy?

andr229k, 2022-01-26:
> What other floppy emulator did you test? Another Gotek? With FlashFloppy?

Hi Keir,

Thank you for the reply.

The Gotek model is SFRC922D, which means that it is a STM32 revision.

keirf, 2022-01-26:
If another Gotek works then this Gotek is clearly at fault.

andr229k, 2022-01-26:
> If another Gotek works then this Gotek is clearly at fault.

I will try get the drive tested with something else

closed 2022-02-19 by keirf