keirf/flashfloppy #601: Can't acces to GOTEK from MS-DOS

8art33z, 2022-01-21:
Would you please let me know how can I boot MS-DOS from a GOTEK which is upgraded with FlashFloppy firmware? I tried many times to format an empty image file and access the image contents from drive A: (which is the Gotek); It tries to access for some seconds. I can see the activity from the LED and the LCD display but eventually it doesn't work and everything stops. Same thing happens during the MS-DOS boot.

davidvrootech, 2022-01-22:
[image]
This is what I get with firmware 4.4a running on the SFRC922AT3. I have tried different flash drives and settings.

keirf, 2022-01-22:
@davidvrootech Problem may be filesystem in the IMG file. Block level problems would give explicit access errors.

8art33z, 2022-01-22:
For me file FF.CFG do the trick. Jumper JC didn't work with my latest "crippled version" of GOTEK.

keirf, 2022-01-22:
Gotek board KC30AT2 does not have a JC jumper unfortunately. You will see JC is not labelled on the PCB.

8art33z, 2022-01-22:
> Gotek board KC30AT2 does not have a JC jumper unfortunately. You will see JC is not labelled on the PCB.

That's it!

closed 2022-01-23 by 8art33z