keirf/flashfloppy #602: Feature request: Support .mmb container files for the Acorn BBC Model B + Master series computers

kgl2001, 2022-01-30:
Hi Kier,

The .mmb file format is a common format used in the BBC Micro series of computers. It is a container file that contains upto 511 200kb single sided disk image (.ssd) files. These .ssd files are already fully supported by flashfloppy. The .mmb file is commonly used with a low cost SD card based drive (and associated MMFS file system ROM) that is connected to the User port of the BBC micro.

I wonder if it would be possible to support the use of these .mmb files in flashfloppy. What I'm thinking is that the .mmb container file could be treated like a directory full of .ssd files, similar to any other directory on the USB stick.

The spec for the .mmb container file is fairly simple. It's detailed here:

https://github.com/hoglet67/MMFS/wiki/MMB-File-Format

Thanks
Ken.

keirf, 2022-01-30:
Unlikely I will support this on the Gotek, but I could supply a script for ripping an MMB into a folder of SSD files. Would that be useful? If so, can you point me at some MMBs?

kgl2001, 2022-01-30:
Thanks for the quick reply. I wouldn't spend too much time on a script. There are already tools to manipulate the MMB file. However, if you do ever want to look at this further, here's a couple of mmb examples:
https://www.dropbox.com/s/yhd8xbunbiqfaa6/BEEB.zip?dl=0
https://github.com/hoglet67/Ice40Beeb/releases/download/release_1/BEEB.MMB.zip

closed 2022-02-19 by keirf