keirf/flashfloppy #603: Write errors with Sony SMC-70

snhirsch, 2022-01-30:
The Sony SMC-70 is one of the first computers to use 3.6" drives.  They are early Sony units that are SSDD, 70-track and spin at 600 rpm.  I was able to get reads working by specifying a 500kbps data rate along with the proper geometry. However, write operation is very flaky.  Sometimes it will appear to succeed, but the data (or directly entry, at least) never appears on the disk.  Other times it will fail with a complaint that the disk is R/O.  And, occasionally, it will succeed.  No rhyme or reason to it.  I am using 'raw' DSK format to avoid overhead from HFE conversion.  The Gotek is an older STM32 based unit.  Any suggestions for getting writes working would be appreciated!

keirf, 2022-01-30:
What is in IMG.CFG? Did you specify `rpm=600` along with `rate=500`?

snhirsch, 2022-01-30:
Yes. Rpm is set at 600 - although, interestingly enough, the behavior is just about identical with either that or the default of 300. Here's the image definition:

```
# Sony SMC-70
[::286720]
cyls = 70
heads = 1
id = 1
interleave = 1
secs = 16
bps = 256
mode = mfm
rate = 500
rpm = 600
```

keirf, 2022-01-31:
Okay, makes sense, default 300rpm would simply leave a lot of track empty space. The IMG.CFG looks okay.

What USB stick are you using? Have you tried others?

snhirsch, 2022-01-31:
I was using a 4GB Sandisk Cruzer, but switched to a 16GB Sandisk Extreme at your suggestion.  No real difference in (mis)behavior.  The LCD on FF shows seeking and writing, but it's not all reliably making it into storage!  For example, it goes through the motion of pip-ing a 20k file from one of the floppies to FF and comes back to the prompt with no errors.  When I list the directory, I can see the file still has the extension of '.$$$'.  So CP/M went the motions of renaming that temporary file to the actual extension and believed it had succeeded - however this change didn't make it to storage.  I'll try a few other sticks, but these have worked just fine on a wide variety of systems.

snhirsch, 2022-01-31:
And... Success!  Whatever the underlying issue was, you seem to have fixed it in the latest pre-release (4.4).  Works just fine and I haven't been able to trip it up yet.  Unless you want me to try anything else for forensic purposes I think we can close this issue.

Once again THANKYOU for your terrific work on FlashFloppy!

keirf, 2022-01-31:
Was the previous failure on v3.x? Or were you using an earlier pre-release?

snhirsch, 2022-01-31:
I am almost positive it was a 3.x release, but am embarrassed to admit I didn't take note of it before upgrading. Things work terrifically now.

keirf, 2022-01-31:
It may make sense. Some hosts are sensitive about write latency and v4 should be a lot better for that.

closed 2022-01-31 by keirf