keirf/flashfloppy #606: "ERROR: Unknown Error ! [-1]" / Gotek SFRC922D with Artery MCU

sandfern, 2022-02-12:
Hi. I recently bought a Gotek to use with my Amiga 500. I'm trying to use it first as an external drive. The Gotek board is the SFRC922D with the Artery AT32F415RCT7 MCU, preconfigured with the 3.29 firmware.

When I try to use it through Workbench, the FF File Selector shows a "ERROR: Unknown Error ! [-1]" message.

I've tried several combinations of usb sticks brands and sizes, plus several combinations of firmware versions (3.29, 3.30 and the experimental 4.4a). As issue #161 mentioned the use of the HxC_Compat_Mode-v5 as a possible solution, I tried it together with those firmware versions, with no results (I also posted there with the hope of someone knowing a fix and so not needing to open a new issue).

I've already contacted the Gotek seller who only said the error was probably due to a mismatch between the firmware and the FF version used and to update the firmware to the same version of FF I was using. I told him the error persisted even after updating the firmware but he still hasn't replied.

Am I doing some basic mistake here or is the Gotek board broken somehow? Thank you for any help!

keirf, 2022-02-12:
Does it work if you boot from it?

sandfern, 2022-02-12:
I haven't tried it yet. I would prefer not to remove the Amiga internal drive, one of the bolt screws (I think that's the name of it) that secures it to the screw accessible from the outside spins without unscrewing itself from the drive and I'm not able to detach it from the screw (I think the only way I have to remove the drive would be to break the case around the screw) so I'll be installing a DF0 Selector to the Amiga's CIA chip but I haven't done it yet.

keirf, 2022-02-12:
Ah is it an older Kickstart which can only boot df0?

sandfern, 2022-02-12:
Yes, it's 1.3.

sandfern, 2022-02-12:
Hmm, Maybe it's possible to connect the Gotek to the Amiga as df0 without physically removing the original drive? I'm going to check it later.

keirf, 2022-02-14:
How much Chip mem and Total mem do you have available when you run the file selector from WB?

sandfern, 2022-02-17:
I'll be able to check it tomorrow. The Amiga has 1MB RAM total though (512 + 512 expansion).

sandfern, 2022-02-17:
Ok, I already checked with the util that comes with the WB extras (Freemap I think?):

Chip mem: 425504
Fast Mem: 420920

The util itself was using some memory but, even when launching the selector right after booting, without launching anything else, I still get the error.

keirf, 2022-02-17:
Should be enough RAM. I will have to test this at some point I think and see if I can reproduce it.

So you boot WB1.3 on disk in DF0, and then have the AUTOBOOT.HFE in Gotek DF1 which you click to start?

sandfern, 2022-02-18:
Yes. WB1.3 in df0, gotek in df1, with AUTOBOOT.HFE (and HXCSDFE.CFG) on the root -> "unknown error -1" after clicking the FFmanager icon (the selector screen appears but after one or two seconds, a box with the error appears).

If the Gotek is installed as df0, no error in v3.30 and everything works as it should, as far as I know.

I tried this with two different WB1.3 versions. One is the official Cloanto version, bought at Amigastore. The other was a bootleg copy that came with the Amiga. It happens with both.

If there's any other information I may provide you with, let me know. I've only started using an Amiga for the first time a few months ago so I'm a complete noob with it but I'll do my best to help with anything you may need.