keirf/flashfloppy #607: Akai S612 and saving problem

andr229k, 2022-02-13:
**Environment:**
Akai S612 with Gotek [FRKC30.AT2] and Flashfloppy QuickDisk firmware version 3.30.

**Symptoms:**
Floppy Emulator loads the Akai Quickdisk Factory Disks without problems. But the following happened by saving.
- Selecting blank disk on Floppy Emulator display
- Pressing save on s612, «d» starts blinking. Pressing save once more and «d» is continuous for some seconds, but not as long as the load time. Press verify on the s612, the display says «d» then blinks «E» for «error» . 

**Diagnose:**
FRKC30.AT2 design does not feature JC pin and it is normally communicated that jumper on M0 and sometimes JC is needed as per following wiki:
- https://github.com/keirf/FlashFloppy/wiki/Quick-Disk
- https://masutos.wordpress.com/2021/02/02/akai-s612-md280-qd-drive-replacement/?fbclid=IwAR0DBI2CdCGg2jY7i9eYrIogboWYrP419Hb-04AmzIOOrM_ahCGAD4YFIJg < Here mentioned that JC is needed, but it was before design change.

**Action: [Various tests]**
**Jumper on MO only, without jumper on Pin 10:**
- Green LED on Gotek lights, but the device is stuck trying to load .QD factory sound disks, same for saving.

**Jumper on M0 only, with jumper on Pin 10:**
- Green LED on Gotek lights, but the device is stuck trying to load .QD factory sound disks, same for saving.

**Jumper on SO only, without jumper on Pin 10:**
•	No Green light by handshake and stuck in loading/saving.

**Jumper on SO only, with jumper on Pin 10:**
- Device is able to load factory sound disks without problems.
- After I loaded a factory disk into the memory, then I wanted to re-save it onto a blank .QD disks. The Gotek LED starts blinking as the saving process is initiated and after a while Akai S612 shows “E” on the screen, which means Error.

**Next:**
Any ideas why im able to load, but not saving?

Note:
- Jumper on Pin 10 was tested due to that it was illustrated on the HxC wire chart. [Attached]
- The missing JC on the design does not seems to be the problem as im able to load and not save?

[image]

andr229k, 2022-02-14:
Logs uploaded [Log and the used blank disk]
[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/8061522/FFLOG.TXT)
[Blank.qd.txt](https://github.com/keirf/flashfloppy/files/8061523/Blank.qd.txt)

### Event:
-Seated jumper SO and on Pin 10
-Tried to load Blank.qd, resulting in «E»
-Tried to load some samples, some loaded good, some returned «E»
-Tried to save a sample on Blank.qd, looks like its working, but «D» stops after only 3 seconds or so (should take 8)
-When pressing VERIFY or LOAD Blank.qd, returns only «E»

andr229k, 2022-02-14:
Another attempt with same setting as above with jumper on S0 and Pin 10

**Event:**
-Loading grand piano sample 
-Saving grand piano on blank - 3 sec
-Verify - returning «G»
-Loading trumpet sample
-Selecting Blank.qd
-Try to load Blank.qd, but only «E» on display

[Blank.qd.txt](https://github.com/keirf/flashfloppy/files/8062640/Blank.qd.txt)
[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/8062641/FFLOG.TXT)

andr229k, 2022-02-15:
> Write seem to get saved okay albeit in the first test case it perhaps wraps past the end of the QD image which isn't good. The second test with the piano sample I can't see anything weird like that though. Most odd.

That a Jumper on S0 and Pin 10 works over a jumper on MO only is also interesting.

Are there any settings or adjustments, which could improve the saving?

Could it be a problem with the image?

With Flashfloppy there is only 1 x .QD image attached, but for HxC there is 4 different Quickdisk images.  [see attachment]
[QuickDisk.zip.txt](https://github.com/keirf/flashfloppy/files/8068879/QuickDisk.zip.txt)

keirf, 2022-02-15:
Yes, the need to use jumper + S0 is indeed also interesting. Not heard of that one before.

Any of the HxC QD images will be compatible and you could certainly try them.

andr229k, 2022-02-19:
> Yes, the need to use jumper + S0 is indeed also interesting. Not heard of that one before.
> 
> Any of the HxC QD images will be compatible and you could certainly try them.

The Floppy Emulator has now HxC and it works without problems with jumper on S0 and pin-10.

Seems like Flashfloppy Quickdisk is not yet compatible with Akai md280 / S612

keirf, 2022-02-19:
Okay so compared with HxC firmware we have:
* Unreliable loading
* Saving never works
Would that be a fair summary?

andr229k, 2022-02-19:
> Okay so compared with HxC firmware we have:
> 
> * Unreliable loading
> * Saving never works
> 
> Would that be a fair summary? Was this using the HxC Roland QD firmware, or the non-Roland regular QD firmware?

My conclusion is that loading works, but saving is an issue.

But, there Is mentioned success in the following two articles: 
https://masutos.wordpress.com/2021/02/02/akai-s612-md280-qd-drive-replacement/
https://github.com/keirf/flashfloppy/issues/509

Both are from 2021, did you change much in regard to Quickdisk around Jul 21 till today?

Im also asking myself, what they tested with their devices, if they both tested loading and saving.

keirf, 2022-02-19:
Really nothing has changed in QD since July last year. Possibilities here:
1. Artery MCU reduced RAM is a problem. Hence worth trying v4.4a to see if that helps.
2. Jumper JC is mentioned in your link and may be required. It would be equivalent to the HxC Roland QD firmware. Is that the HxC firmware you used?

Your link mentions loading and saving so I think we can assume the ARM-based Gotek worked okay for them.

TheSunGodRA, 2022-02-19:
Hi. I am the owner of the Akai S612 / MD280 with the Gotek drive being discussed here. Bought a FF-preflashed drive from Andr229K, being advertised as "plug & play" and "Fully compatible" with Akai MD280, among others. As this thread with my results shows, that was hoping too much. The seller was kind enough to refund so that I could install HxC firmware instead. Now, with jumpers on "MO" (not SO) and pin 10, everything works as it should. I am therefore not able to test FF on it any longer.

keirf, 2022-02-19:
Thank you for the update!

closed 2022-02-19 by keirf