keirf/flashfloppy #608: Akai s900 should say S1 "host platforms"

ilantz, 2022-02-14:
I've followed the note to set the jumper to S0 for S900, but accessing the disk feature on the akai triggered the "OOPS! Disk not mounted." error. setting the jumper to S01 solves it. 

Tot a Gotek SFRKC30.AT2, running fp 3.30, and an Akai s900

keirf, 2022-03-10:
Which note is this? https://github.com/keirf/flashfloppy/wiki/Host-Platforms#akai states that the jumper position is usually S0 for all Akai, and there is no further mention of jumper setting for S900 specifically.

ilantz, 2022-03-10:
Right, but in there, the heading/title of "S01, S900:" had me understanding that it should be S01 for S900... perhaps would be worth to just remove "S01, "

keirf, 2022-03-10:
But SO1 is legitimately an Akai model.

ilantz, 2022-03-10:
🤦‍♂️ gotcha. complete oversight on my part.

closed 2022-03-11 by keirf