keirf/flashfloppy #611: The Gotek driver can't work with EPSON PC-486SE /PC9801 ,firmware version is v1.2

leothebest2014, 2022-02-21:
Hello, sorry to bother you! I have a EPSON 486, the original floppy driver is SMD340-301 which has 26 pin    
laptop 1.25mm fpc interfaces . The problem is that the computer doesn't recoganize the Gotek, when I try to operater the floppy on the computer, the computer tells me that the floppy isn't ready. The pinout of the adapter is as follows:
(26-pin) -> (34-pin + power)
1 -> 5V
2 -> 8
4 -> 10
6 -> 2
7 -> 12
8 -> 34
10 -> 16
12 -> 18
14 -> 20
15 -> GND
16 -> 22
18 -> 24
20 -> 26
22 -> 28
24 -> 30
26 -> 32

and the status of the Gotek is also right,when a usbstick insert into the floppy ,the 7-seg led showed '000'.At last ,I changed the FF.CFG as follows:
host=pc98
pinout=rdy

still don't work. After that I tried the v2.21 firmware. Don't work. Seems no related.
 I searched all the issues , find no similar issue like mine. Could you help?
[image]
[image]
[image]
[image]
[image]

keirf, 2022-02-21:
Are you sure the adapter is correct?

leothebest2014, 2022-02-21:
I' sure ,I tested every connection one by one.

keirf, 2022-02-21:
Try `interface=ibmpc` in FF.CFG and get rid of the `pinout=` line as it is not a valid option and so just confusing. Are you sure `host=pc98` is correct? This will cause IMG files to be recognised as Japanese-style PC98 360rpm floppies.

leothebest2014, 2022-02-21:
Ok, I'll try, the image file format should be ok,because last time this gotek floppy with version 1.2 worked ok in another pc98 computer with 34 pin interface. So I   
assume the format should be ok!

keirf, 2022-02-21:
If the physical connection is okay, and the image is okay, then it would probably be the `host=` or `interface=` settings. Assuming the laptop itself is okay!

leothebest2014, 2022-03-09:
Hi,can you give me a adapter' link on ebay, I don't know the name, so I can't find it. I want to buy one to try,

keirf, 2022-03-09:
The more common 26-pin interface is different than yours: https://old.pinouts.ru/HD/26pin_microfloppy_pinout.shtml

Hence I don't know about compatible off-the-shelf adapters.

leothebest2014, 2022-03-10:
I find the internet and finally find the pinout of epson smd 340-301 model floppy driver. According to this document , the 7 pin of smd 340 is 5V, but in your  wiki , the 7 pin of 26 interface is connected to 12 of 34 pin. Is there any possibility that cause the problem?
[image]

keirf, 2022-03-10:
Pin 12 of the Gotek isn't actually connected to anything unless you jumper position S1. Since you probably jumper S0, it should have no effect.

leothebest2014, 2022-03-10:
OK， I SEE，I'LL TRY OTHER WAYS.